node-opcua/node-opcua #1380: Documentation is not working

Rednas83, 2024-07-22:
**Describe the bug**
After folllowing [creating_a_client_typescript](https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/creating_a_client_typescript.md) I am getting an error
[image]

Tried
* Importing from `node-opcua-client` instead
[image]
* Installing `node-opcua`
[image]

In both attempts `ReadValueIdLike` is missing which seems to be required for subscriptions.

Also I am getting a deprecrated warning when using `readVariableValue`
[image]

After removing 'ReadValueIdLike' I get the result as expected
[image]

How can you browse the objects inside a folder structure?

Something like
[image]

Can you update the manual with the correct setup procedure?

**Expected behavior**
Working documentation

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- (x) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - (x) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:
EcoStruxure OPC UA Server Expert v2.1
[image]

- **node-opcua version:** :
```
  "dependencies": {
    "node-opcua-client": "^2.129.0"
  }
```
- **Node:**
[image]

Rednas83, 2024-08-02:
Any updates on this?

Also I was missing in the documentation how to read the discovered variables.

Something like
`const items = await session.ReadVariables() // returning [id, name, users, etc]`

How can this be achieved?

erossignon, 2024-08-02:
You  can find tutorial and more in-depth examples in my book at https://leanpub.com/node-opcuabyexample-edition2024

closed 2024-08-02 by erossignon

Rednas83, 2024-08-02:
Why did you close? The documentation is still outdated

Figured it out myself how to get the endpoints
`const endpoints = await OPCUAClient.getEndpoints()`

Rednas83, 2024-08-05:
Just tried it but it only returns the server and not its content.
[image]
How to return the discoverable variables of the server?

Something like
[image]

Is it supported? I am not able to find it in the docs or in the examples at https://github.com/node-opcua/node-opcua/tree/v2.1.3/documentation