node-opcua/node-opcua #1382: Add support for ecmascript module to support import

Rednas83, 2024-07-25:
I am getting
[image]

Can you also make the module available as an ecmascript module to support `import`?

It was introduced in 2015 and has become pretty much become the standard for modules.

erossignon, 2024-08-01:
Some prototyping work has been done in this regard for node-opcua-crypto and node-opcua-pki as a proof of concept. 

Now waiting for donateur to commit development effort on the 72 modules of node-opcua

see https://opencollective.com/node-opcua/projects/convert-to-esm-module

Rednas83, 2024-08-02:
Thanks for your response.

Is it not better to get income by simply applying the library for customers with a need of event driven communication? It's a lot faster then the conventional way by polling data so use cases should be easily to find.

**For example**
* Logger
* Simulation
* Browser via data discovery

**Advantages**
* Improve by learning from actual projects
* Earn income for improving the library without relying on sponsorships
* Faster growth of users without scaring users with sponsorships
* Faster development
* Most common approach for open source projects

erossignon, 2024-08-02:
>Is it not better to get income by simply applying the library for customers with a need of event driven communication? It's a lot faster then the conventional way by polling data so use cases should be easily to find.

OPC UA allows you to have even driven communication by design. Check out subscription and how they work. 
Polling data out of a OPCUA server by recurring reads is a bad design practise.

Rednas83, 2024-08-04:
I agree. Unfortunately polling is pretty much the default within OT.

Libraries like this could help the adoption of OPC UA but RPC is also getting adopted more and more.

closed 2024-11-27 by erossignon

erossignon, 2026-09-14:
node-opcua@2.184.0 onward is a "module" ,