node-opcua/node-opcua #1384: How to browse the discoverable tags from an endpoint

Rednas83, 2024-08-06:
**Current behavior**

**Describe the bug**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. ....
2. ....
3. ....

**Expected behavior**

**Screenshots**

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :

  
  

- **Node:**

  `node --version` =

AndreasHeine, 2024-08-06:
Example how to Browse:

https://github.com/AndreasHeine/opcua-machinery-client/blob/728cc23c0ecf1ee5de17e45a2bed9d50762e8acc/src/ua-device.ts#L826

Browsing is following References! 

```ts
    private async findMachinesOnServer() {
        console.log(`OPC UA Client: findMachinesOnServer...`)
        const machineryIndex = this.getNamespaceIndex("http://opcfoundation.org/UA/Machinery/")
        if (machineryIndex === undefined) return
        const machinesFolderNodeId = `ns=${machineryIndex};i=1001` // id is defined in spec. and can be hardcoded!
        const browseResult = await this.session!.browse({
            nodeId: machinesFolderNodeId,
            browseDirection: BrowseDirection.Forward,
            referenceTypeId: ReferenceTypeIds.Organizes
        } as BrowseDescriptionLike)
        if (isStatusCodeGoodish(browseResult.statusCode)) {
            browseResult.references!.forEach((result) => {
                this.foundMachines.add(makeNodeIdStringFromExpandedNodeId(result.nodeId))
            })
            console.log(`OPC UA Client: found '${this.foundMachines.size}' machine instances!`)
        }
    }
```

so in the example above i have a nodeid where i want to start (may be "Root" or "Objects" or another common node which every OPC UA Server has). browseResult ("BrowseDescriptionLike") contains the found references ("Nodes" / Expanded-NodeId's) for the specified browseDirection ("Forward") and referenceTypeId ("Organizes") in this case it returns all Nodes which got "Organized" by the "machinesFolderNodeId"

References:
https://reference.opcfoundation.org/Core/Part4/v104/docs/5.8.2
https://reference.opcfoundation.org/Core/Part3/v104/docs/4.3.1#_Ref114466674

Rednas83, 2024-08-06:
Thanks for sharing this project. It looks very intersting.

I would like to browse all variables inside "Objects/.ePAC:Project"
[image]

Based on your project I Tried
```
    console.log(`OPC UA Client: findMachinesOnServer...`)
    const machinesFolderNodeId = `ns=2;s=Objects` // id is defined in spec. and can be hardcoded!
    const browseResult = await session!.browse({
      nodeId: machinesFolderNodeId,
      browseDirection: BrowseDirection.Forward,
      referenceTypeId: ReferenceTypeIds.Organizes,
    } as BrowseDescriptionLike)
    console.log(browseResult)
```
But it returns
[image]

Any idea how to do this? The namespace is `2` and the folder I am trying to read is `Object/ePAC:Project`

AndreasHeine, 2024-08-06:
`ns=2;s=Objects` is not a valid NodeId of your Server!

AndreasHeine, 2024-08-06:
[image]

Rednas83, 2024-08-06:
Can you help me setting up the `NodeId`?
Hereby the node attributes of the folder that I am trying to 'browse'
[image]

Tried
```
ns=2;i=0 // Returns 'The node id refers to a node that does not exist in the server address space.'
2:ePAC:Project // Returns 'String cannot be coerced to a nodeId : 2:ePAC:Project'
2:Objects // Returns 'String cannot be coerced to a nodeId : 2:Objects'
ns=2;i=85 // Returns ''The node id refers to a node that does not exist in the server address space.''
```

Hereby the node attributes of the folder I am trying to 'browse'
[image]
[image]

AndreasHeine, 2024-08-06:
just as an explanation:

"ns=2;i=0" translated to OPC UA means NamespaceIndex: 2 IdentifierType: Numeric ("i") and Identifier: 0
"2:ePAC:Project" is actually a BrowseName and not an NodeId...
"2:Objects" same... but Objects is always in NamespaceIndex: 0
"ns=2;i=85" translated to OPC UA means NamespaceIndex: 2 IdentifierType: Numeric ("i") and Identifier: 85 (85 is the Identifier of Objects but not in Namespace 2!)

Rednas83, 2024-08-06:
I missed the ":" in the end.

I now get a good result.
[image]

Now I probably need to map the `reference `to `reference.browseName.toString()?`

AndreasHeine, 2024-08-06:
now you know the NodeId's of all childnodes with that ReferenceTypeId!

its like you opend a folder in your filesystem... now you can Browse deeper or read some Attributes of the found Nodes... whatever your application should do ^^

Rednas83, 2024-08-06:
I managed to get the names by mapping the `reference` to `reference.nodeId.value`

It returns something like
[image]

Any idea what the prefix of '0:' means?

AndreasHeine, 2024-08-06:
thats beyond volentary/free support! you need to understand some of the core concepts of the OPC UA AddressSpace first... your questions are very basic... and i guess you need a crashcourse in OPC UA 😅

Rednas83, 2024-08-07:
Thank you for support on browsing the variables inside Objects. I indeed would consider this as basic functionality. Perhaps it can also be added it in the documentation/example?

AndreasHeine, 2024-08-07:
there is no need of extra documentation... it works like the OPC UA spec. describes it!

here you can start:
https://reference.opcfoundation.org/Core/Part1/v105/docs/6.3
https://reference.opcfoundation.org/Core/Part1/v105/docs/5.4.2
https://reference.opcfoundation.org/Core/Part1/v105/docs/7
https://reference.opcfoundation.org/Core/Part3/v105/docs/4.4
https://reference.opcfoundation.org/Core/Part4/v104/docs/5.8.2

Rednas83, 2024-08-07:
A specification is usefull for writing a library. But not very practical
for creating an application that needs to use this library.

Please consider adding an example. I think it will be appreciated.

AndreasHeine, 2024-08-07:
already availble... https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client_typescript.md#browsing-the-root-folder

but even with plenty examples... you need to know the basics to start building your OPC UA Application... i know where your at, i was there too 😉 skipping the basics creates a lot frustration while try and error!

Rednas83, 2024-08-07:
There is an example for browsing the root folder
[image]
But for others folders you would need to find out the NodeId first.

Something like:
`const TAG_PREFIX = "ns=2;s=0:" // BME NUA 0100 - Requires namespace (ns=2) and objects folder (s=0)`

Why is not possible to browse other folders by string just like "RootFolder"?

I don't mind if it's added or not but I think it could be helpfull for others in the community.

AndreasHeine, 2024-08-07:
> Why is not possible to browse other folders by string just like "RootFolder"?

not part of the OPC UA Spec.!

there is a BrowsPath-Translation-Service (which you have to know exists):
https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client_typescript.md#finding-the-nodeid-of-a-node-by-browse-name

but for that you need to know what:
Namespaces, NodeIds and Browsnames / BrowsePaths are (and BrowsePaths have a bitter flavour... because they require you to apply the correct NamespaceIndex to the BrowseNames!)

there is no shortcut (longterm)...

Sample-Server with Sample-Client (build for www.umati.org):
https://github.com/AndreasHeine/opcua-machinery-client
https://github.com/AndreasHeine/SampleServer-node-opcua

Rednas83, 2024-08-10:
That's not looking very performant🤔

I prefer to hardcode the `nodeid`.

Thanks for your support

closed 2024-08-10 by Rednas83