node-opcua/node-opcua #1385: Return session returns circular reference error

Rednas83, 2024-08-14:
I have a simple `connect` function and I would like it to return the session. But I am getting
[image]

Is there a way to avoid this error?

**plc.server.ts**

```
var session: ClientSession, tags: string[], subscription: ClientSubscription
export async function connect(): Promise<ClientSession | undefined> {
  // 1 Create a client
  const client = OPCUAClient.create(options)

  // 2 Connect to the server
  // TODO Error handling
  try {
    await client.connect(URL.SIMULATOR)
  } catch (error) {
    console.log(error)
    if (error instanceof Error && (error.message.includes("ETIMEDOUT") || error.message.includes("ECONNREFUSED"))) {
      // return { time: time(), type: "error", value: `PLC is offline`, trace: `plc.connect(${URL.SIMULATOR})`} as log
      // return `e-PLC is offline t-plc.connect(${URL.SIMULATOR})` // TODO Add support for string
      return undefined
    } else {
      // return {
      //   error: {
      //     time: time(), // ISSUE Composables are not supported on the server?
      //     type: "error",
      //     value: `Endpoint is invalid! Only ${await client.getEndpoints().then((item) => item.toString())} are available`,
      //     trace: `plc.connect(${URL.SIMULATOR})`,
      //   } as log,
      // }
      // throw createError(
      //   `e-Endpoint is invalid! Only ${await client.getEndpoints().then((item) => item.toString())} are available t-plc.connect(${URL.SIMULATOR})`
      // )
      return undefined
    }
  }

  // 3 Browse the tags
  session = await client.createSession()
  const browseResult = await session.browse({
    nodeId: TAGS_FOLDER.SERVER,
    browseDirection: BrowseDirection.Forward,
    referenceTypeId: ReferenceTypeIds.Organizes,
  } as BrowseDescriptionLike)
  tags = browseResult?.references?.map((item: any) => item.nodeId.value) || []

  // 4 Simulation the inputs by subscribiting to the outputs
  console.log(new Date(), `Connected to ${session.endpoint.endpointUrl} and found ${tags.length} tags`, session.endpoint.endpointUrl)
  subscription = ClientSubscription.create(session, {
    requestedPublishingInterval: 1,
    requestedLifetimeCount: 100,
    requestedMaxKeepAliveCount: 10,
    maxNotificationsPerPublish: 100,
    publishingEnabled: true,
    priority: 10,
  })
  // await simulate(tags.filter((item) => OUTPUTS.some((value) => item.endsWith(`_${value}`))))
  return session
}
```
**Version**
```
------------------------------
- Operating System: Windows_NT
- Node Version:     v20.12.2
- Nuxt Version:     3.12.4
- CLI Version:      3.12.0
- Nitro Version:    2.9.7
- Package Manager:  pnpm@9.7.0
- Builder:          -
- User Config:      telemetry, modules, experimental, nitro, icon, build, routeRules, components, content, formkit, imports, css, devtools, compatibilityDate
- Runtime Modules:  @vueuse/nuxt@10.11.1, @nuxt/devtools@1.3.9, normalizedModule(), @nuxt/content@2.13.2, @formkit/nuxt@1.6.5, @nuxt/image@1.7.0, @nuxt/ui@2.18.4, nuxt-remote-fn@0.5.3
- Build Modules:    -
------------------------------
```

AndreasHeine, 2024-08-14:
classes with methods can not be serialized as JSON...

Rednas83, 2024-08-14:
Is it not possible to just return the handle of the session? It would be much more convenient for exporting/sharing I think🤔

Something like?
```
  // Return a safe representation of the session
  return {
    sessionId: session.sessionId,
    endpointUrl: session.endpoint.endpointUrl,
    subscriptionId: subscription.subscriptionId,
    // Add other necessary properties here
  } as unknown as ClientSession;
```

AndreasHeine, 2024-08-14:
unfortunately its not that easy...
a session can have multiple subscriptions!
and of course the binding to the ClientClass and the Transport... its not just a handle...

unfortunatly there is no toJSON method in the ClientSession class..

but if you implement it you better stick to the Interfaces of the Classes just remove the methods and other non serializable stuff!

e.g.

https://github.com/node-opcua/node-opcua/blob/af91638471767c90059a973cdc969b903eff0953/packages/node-opcua-client/source/client_session.ts#L100
https://github.com/node-opcua/node-opcua/blob/af91638471767c90059a973cdc969b903eff0953/packages/node-opcua-client/source/client_session.ts#L133

Rednas83, 2024-08-14:
Thanks for the feedback.

Perhaps something for the roadmap?
- [ ] ClientSession.toJSON()
- [ ] toClientSession()

Btw for me it's unclear why it's trying to convert to JSON in the first place. Why is it not kept as a class? Perhaps it could also be an issue with Nuxt

AndreasHeine, 2024-08-14:
just as clearification node-opcua does not work in the browser it uses nodejs sockets which browsers cant access so everything with regards to node-opcua has to be in the backend!

i have seen no usecases so far to revert JSON to a ClientSession-Class sounds wild ^^

Rednas83, 2024-08-14:
Or perhaps something like a connect via a handle
`var session = await client.getSession(123)`

AndreasHeine, 2024-08-14:
the clientclass did not store its created sessions, the client-class-instance is injected into the session-class-instance instead

AndreasHeine, 2024-08-14:
"connect via a handel" not sure what your aiming at!?

OPC UA Spec defines the whole procedure:
[image]

AndreasHeine, 2024-08-14:
@Rednas83 

just a wild guess you experience Session exhaustion from your connect method been called to often (the reason why you use sessiontimout 1000ms) always creating new sessions and dont let node-opcua reuse the existing?

looking in your code snippet:

```
requestedPublishingInterval: 1,
```
are you serious about requestedPublishingInterval? -> 1ms !!! this will be revised by the PLC for sure!

do you plan to DoS the PLC ^^ 

to small publishing intervals are dangerous because if you have slow datachangerates e.g. over the weekend (Machine Off) the Subscription will die because of the Lifetimecount or Keepalivecount!

use longer PublishingIntervals for example 500ms/1000ms and smaller SamplingIntervals e.g. 100ms / 250 ms (usually the PLC will revise the requested values if your values are not applicable)

[image]

i wrote a aricle about Subscriptions/MonitoredItems a while ago because many of my customers struggle with the settings -> https://www.linkedin.com/pulse/brief-introduction-opc-ua-subscriptions-andreas-heine-3pc6e/?trackingId=Pu%2FyWa8ynTNQqCLPnw%2B3Og%3D%3D

Rednas83, 2024-08-14:
Pointers/handlers are not supported in nodejs
[image]
And references will only work within the scope.

I will look for a way to make the session available globally so that I don't need to pass the `session` around. Probably that is easiest for now.
**Advantage** No need to return/share the session
**Disadvantage** This will probably also inject it in all my pages on client level

Can you please consider support for sharing/exporting sessions in an easy manner?

Thanks for pointing out `requestedPublishingInterval`. Schneider M580 should be capable of 2000 subscriptions in 20ms
[image]

I just wanted it to be asap but I now modified it to 20ms which should be okay.
`requestedPublishingInterval: 20,`

AndreasHeine, 2024-08-14:
> I will look for a way to make the session available globally so that I don't need to pass the session around. Probably that is easiest for now.
Advantage No need to return/share the session
Disadvantage This will probably also inject it in all my pages on client level

i guess you will endup in big trouble with global session object approach! if you look into my example client you will have the base structure for a OPC UA Device Proxy Class 
https://github.com/AndreasHeine/opcua-machinery-client/blob/7388b633e2db69496f3f7bcf9a1cf84541b8af37/src/ua-device.ts#L134

sure you need to delete a lot of the stuff but you will have a usable object oriented approach.

> Thanks for pointing out requestedPublishingInterval. Schneider M580 should be capable of 2000 subscriptions in 20ms

your mixing the terms... 
2000 MonitoredItems not Subscriptions (they are different things!) -> usually PLC's support less then 20 Subscription's each Session (less then 100 in total) and the number of connections/sessions is limited as well!

each OPC UA Server provide that in the Addressspace as well:
[image]

and PublishingInterval is not the same as SamplingInterval!

PublishingInterval just empies the MonitoredItem's queues and transfer them in a batch to the client
SamplingInterval defines how often should the PLC check the Data-Source and apply the filters to it in order to put notifications in the queue!

thats the most important part if Subscriptions and MonitoredItems to understand first, next is Lifetimecount- and Keepalive-Mechanism -> its the number one reason why subscriptions die unwanted... just by wrong parameters... node-opcua calculates the parameters if you leave them open, thats why i dont specifie them in my reference client:
https://github.com/AndreasHeine/opcua-machinery-client/blob/7388b633e2db69496f3f7bcf9a1cf84541b8af37/src/ua-device.ts#L114

for plc connections i used (worked >90% sufficent):
30s-360s SessionTimeout
1000ms PublishingInterval (for most applications fast enought its just the transport delay for the batch)
and 100m / 250s (depends on the plc model) SamplingInterval for Analog-Values
and 1000ms SamplingInterval for Parameters

Queue-Size you can calculate as well:
1000ms PublishingInterval with SamplingIntercal 100 -> max 10 Changes in 1000ms  would be Queuesize 10 (i prefer to set it to 5x the min Queuesize means 50)

AndreasHeine, 2024-08-14:
@Rednas83 just a article i found according to the global object topic:https://www.linkedin.com/pulse/global-variables-dependency-injection-ray-paseur/

Rednas83, 2024-08-14:
My usecase is simulation.
* Subscribe to 200 outputs (boolean) hopefully at the cycle time of the plc of 20ms🤞
* And write back the inputs (boolean)
* I will also do some reading and writing with forms (2000 variables) but that is just simpe read and write depending on the active page/form.

If ua is to slow I will move the simulation to the PLC and keep the forms in the external application but prefered is to have the simulation with external software which can hopefully run at the cycle time of the PLC.

AndreasHeine, 2024-08-14:
did something similar with Emulate3D's UA CLient and a Siemens S7... from that u can tell you you will not reach cycletime! realisticly 100-200ms!
for these kind of hard realtime stuff you need TSN and UADP (OPC UA Pub/Sub)

i would do batch read and write (no subscription) according to the opperational limits of the plc (MaxNodesPerRead / MaxNodesPerWrite)

and then i would mimic the PLC Cycle (but from external view) in the ua client, but add behavieour ("peripheral behaviour" e.g. Valves open and close if the output is set) between Read and Write,
you will not get faster then that!

similar to how WinMOD (https://www.winmod.de/english/) does it simulating Profinet Devices for example!

Rednas83, 2024-08-14:
For siemens I would recommend RPC or MQQT
https://www.youtube.com/watch?v=d9EX2FixY1A
Unfortunately this is not supported for Schneider😢

AndreasHeine, 2024-08-14:
true S7 is limited in its Samplingtask's (the Sampler is the Bottleneck!)

A Client can have multiple Sessions (Server.ServerCapabilities.MaxSessions) a Session can have multiple Subscriptions (Server.ServerCapabilities.MaxSubscriptionsPerSession but not more then Server.ServerCapabilities.MaxSubscriptions) each Subscription can have many MonitoredItems (Server.ServerCapabilities.MaxMonitoredItemsPerSubscription) each MonitoredItem has a own SamplingTask with an interval checking for changes according to the filter

generally speaking the fastest way to get data is a simple "BatchRead":
Session.Read([<Server.ServerCapabilities.OperationalLimits.MaxNodesPerRead>]) like 1k (depends on the plc) in one request

a subscription can be more efficient on the wire but is slower in reactiontime accordint to the longpolling-mechanism (PublishInterval of Request/Response) thats why i recomend "BatchRead" for fast responsetime

AndreasHeine, 2024-08-14:
> For siemens I would recommend RPC or MQQT https://www.youtube.com/watch?v=d9EX2FixY1A Unfortunately this is not supported for Schneider😢

MQTT on S7 suck! just for fun try JSON payload ;)

Rednas83, 2024-08-14:
> > For siemens I would recommend RPC or MQQT https://www.youtube.com/watch?v=d9EX2FixY1A Unfortunately this is not supported for Schneider😢
> 
> MQTT on S7 suck! just for fun try JSON payload as subscriber ;)

I was able to put everything in 2 messages with raw data with very high performace. But that will of course require decrypting on the client side

Rednas83, 2024-08-15:
I saw benchmarks where ua is 5-7 times faster then polling.
So if you would be able to reach 500ms with polling you would get something like 100ms with ua.

AndreasHeine, 2024-08-15:
Subscription works under the hood still with Request/Response (it called "Long-Polling") which is different to MQTT Subscription for example which is a TRUE Pub/Sub!

Best response time/behavior is with good old Read (Multiple Nodes at once) that fact... 
OPC UA Subscription is very efficient because of less Requests and Batchtransfer of ChangeNotifications...

If you look on what happens on the wire:
you want best response-time -> Read/Write
you want efficiency -> Subscription (does not mean its fast ^^)

If you twist some settings in the Subscription e.g. PublishingInterval (Not SamplingInterval!) like less then 100ms you basically sent 10 PublishRequests per Second (Worstcase!).
You need to Cache the results to get a current State of all your variables and you have to manage the Cache in order to get consistency in case the subscription/connection breaks or a bad statuscodes occurs... it adds a lot of complexity to the system... not worth in my eyes!

And i am talking about the PublishingInterval so just the Transfer of Notifications from Server to Client... there is more which can make responsetime slower then read e.g. Sampling:

[image]
https://reference.opcfoundation.org/Core/Part4/v104/docs/5.12.1

So even if a Change has been detected by the Sampling, and the DataChangeNotification has been added to the queue it has to wait till the next PublishRequest (some times there is already a PublishRequest waiting but if you have many MonitoredItems e.g. >1k its getting rarer)

That's knowledge i gained in real plants with real machines and real hardware. Benchmarks are kind of weird because they assume to much and are mostly far from reality.

Rednas83, 2024-08-15:
Only with an external ua server there is polling. In my case the PLC is the ua server which looks every 20ms with a dedicated core for changes in the cache when there is an active subscription. (siemens requires load balancing becauses it uses 1 core for communication and logics making it super slow)

I am not sure how the UA Client communicates with the UA Server but polling seems like a very bad practice because nodejs is event based and therefore really bad in cyclic events.

This is my assumption how it works whereby browse is just a startup event to exchange the variables
[image]

AndreasHeine, 2024-08-15:
from: https://reference.opcfoundation.org/Core/Part1/v105/docs/7.11

[Subscriptions](https://reference.opcfoundation.org/search/6?t=Subscriptions) are entities that periodically publish [NotificationMessages](https://reference.opcfoundation.org/search/6?t=NotificationMessages) for the [MonitoredItem](https://reference.opcfoundation.org/search/6?t=MonitoredItem) assigned to them (see [7.9](https://reference.opcfoundation.org/Core/Part1/v105/docs/?r=_Ref101338661)).

**periodically publish**

AndreasHeine, 2024-08-15:
you might call it "bad practice" but its not changing the fact that OPC UA is designed in <2008 (if i remember correctly) and at that time there was no true Pub/Sub (e.g. MQTT) available (or "mainstream" on the shopfloor) so the rely on whats available at that time and the best way to get some sort of event-notification-mechanism was "Long-Polling"!