node-opcua/node-opcua #1388: Error when calling constructExtensionObject on an object that has a nested ExtensionObject

champloo, 2024-09-11:
**Current behavior**

I am writing an OPC-UA client using node-opcua v2.130 and when calling `constructExtensionObject` on an object that has a nested `ExtensionObject` an error occurs _"Cannot find schema for simple type Structure"_ with the below code. 

See [Structure_error.txt](https://github.com/user-attachments/files/16958250/Structure_error.txt)

The top level extension object `jobOrderEO`, is a [ISA95JobOrderDataType](https://reference.opcfoundation.org/ISA95JOBCONTROL/v200/docs/6.3.4) (node id `ns=4;i=3008`), while the nested extension object, `customParametersEO`, is my custom data type.

```js
const customParameters = {
  param1: "A",
  param2: 2,
  param3: [{
    subparam1: "C",
    subparam2: "D",
  }
};

let customParametersEO = await session.constructExtensionObject(coerceNodeId("ns=8;i=3009"), customParameters); 

const jobOrder = {
  jobOrderID: "Job 1",
  workMasterID: [{
    ID: "ExampleProduct",
  }],
  jobOrderParameters: [{
    ID: "JobOrder",
    value: new Variant({
      dataType: DataType.ExtensionObject,
      value: customParametersEO,
    })
  }] 
};

let jobOrderEO = await session.constructExtensionObject(coerceNodeId("ns=4;i=3008"), jobOrder);
```

Full code available at https://gist.github.com/champloo/0d81d149a538f90fd58cf6cb8624b61a

However note that the server being used is proprietary and I cannot post it here, but noting that it is written in .NET

**Expected behavior**

Was expecting that the object would be created without error.

champloo, 2024-09-11:
I was able to construct the object using a different method as follows

1. Not specifying the nested extension object when creating `jobOrder` pojo. See the commented out section in code below
2. Constructing `jobOrderEO` extension object and then after that assigning the `customParametersEO` to the `value` field

```js
const jobOrder = {
  jobOrderID: "Job 1",
  workMasterID: [{
    ID: "ExampleProduct",
  }],
  jobOrderParameters: [{
    ID: "JobOrder",
    /*value: new Variant({
      dataType: DataType.ExtensionObject,
      value: customParametersEO,
    })*/
  }] 
};

let jobOrderEO = await session.constructExtensionObject(coerceNodeId("ns=4;i=3008"), jobOrder);

jobOrderEO.jobOrderParameters[0].value = new Variant({
  dataType: DataType.ExtensionObject,
  value: customParametersEO,
});
```
While this works, it feels like a workaround. Should the original code work?

champloo, 2024-09-11:
I've also seen the following way used quite a bit to construct extension objects  especially in node-red-contrib-opcua, e.g. https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/opcua/102-opcuaclient.js#L964-L967

```js
let eo = await session.constructExtensionObject(coerceNodeId("ns=4;i=3008"), {});
Object.assign(extensionobject, pojo);
```

However when using an object constructed this way in a method call causes an error "encodeArrayOrElement: object field workMasterID has no encode method and encodeFunc is missing"

erossignon, 2025-04-07:
There are numerous of NodeOPCUA Servers implementing Machinery Job Management successfully.
THis issue is now outdated and has been probably implicitly resolved over time. 
Let close.

closed 2025-04-07 by erossignon