node-opcua/node-opcua #1391: Importing Machinery Job Management not work

Kantiran91, 2024-09-30:
**Current behavior**

The server crash at the import of Machinery Job Management

**Describe the bug**
 As you see we got two warnings with the ISA-95 Job Control and after that an error at the importing. It seems that the use of the ISA95Parameter in the Arguments of the method cased the problem

[image]

[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:
```
import asyncio
import os
from asyncua import ua, Server
from asyncua.common.instantiate_util import instantiate

BASE_DIR = os.path.dirname(os.path.dirname(os.path.abspath(__file__)))

async def import_xml_file(server, file_path, strict_mode=True):
    try:
        await server.import_xml(file_path, strict_mode=strict_mode)
        print(f"Successfully imported {file_path}")
    except Exception as e:
        print(f"Failed to import {file_path}: {e}")
        
async def import_models(server):
    print("Importing companion spec. XML...")
    xml_files = [
        ("models/Opc.Ua.Di.NodeSet2.xml", False),
        ("models/opc.ua.isa95-jobcontrol.nodeset2.xml", False),
        ("models/Opc.Ua.Machinery.Jobs.Nodeset2.xml", False),
        ("models/Opc.Ua.Machinery.NodeSet2.xml", False),
        ("models/opc.ua.glas.v2.nodeset2.xml", False)
       ]
    

    for rel_path, strict_mode in xml_files:
        # Construct the full file path, ensuring OS compatibility
        file_path = os.path.dirname(rel_path)
        await import_xml_file(server, rel_path, strict_mode)

async def main():
    # Erstelle einen OPC UA Server
    server = Server()
    await server.init()

    # Setze die URL für den Server (z.B. localhost)
    server.set_endpoint("opc.tcp://localhost:4840")
    
    # Setze den Servernamen
    server.set_server_name("example")
    
    await import_models(server)
   
   
    # Starte den Server
    async with server:
        print("OPC UA Server läuft! Drücke Strg+C zum Beenden.")
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())

```

**Expected behavior**

**Gist**

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: pc
- OS version: _____11_

    - (x) Windows : version : 11
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- **node-opcua version:** :
2.60.0

- **Node:**

  `node --version` =v19.8.1

Kantiran91, 2024-09-30:
I tested the issue and it seems that there is a problem with the sequence of the imported nodeset. 
The machinery job management seems to imported before the isa95 job control

patrickl92, 2025-01-09:
The issue with the ISA95ParameterDataType is that it contains a property called Subparameters, which has the same data type. This is used to nest parameters:
[image]

However, this design causes a problem when loading the nodeset with the node-opcua library. During parsing of a nodeset, when the library encounters a data type, it gathers the information of all properties within that data type. Once done, the data type is added to the internal list of known data types.
This is usually not an issue, because the data types of the properties have all been processed before. But when trying to process the ISA95ParameterDataType, the library is faced with a problem: One of the properties has the same data type that is currently processed. Since processing has not yet finished, no information about this data type can be retrieved. This causes the following warning:

```
08:11:45.205Z :structured_type_schema        :36    Cannot figure out field category for  {
 ...                                                  name: 'Subparameters',
 ...                                                  fieldType: 'ISA95ParameterDataType',
 ...                                                  isArray: true,
 ...                                                  switchBit: 2
 ...                                                }
```

The ISA95ParameterDataType is then still processed and added to the internal list, but compared to other data types, some information is missing. One consequence I encountered is that it is not possible to use `constructExtensionObject` to construct an ISA95Parameter that contains an extension object as a value, as described here: #1388 
It may also be related to #1393

The same also applies for the ISA95PropertyDataType:
[image]

Please note that the description above is a simplification of the actual nodeset loading process so it is easier to understand.

AndreasHeine, 2025-01-09:
has been fixed in latest version!

erossignon, 2025-01-30:
Closing as fixed in [node-opcua@2.139.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.139.0) and aboce

closed 2025-01-30 by erossignon