node-opcua/node-opcua #1400: Saving a node to file and rewriting it

GabrieleFerronato, 2024-12-13:
hello there!

I'm trying to develop some code that allows me to read a node that contains a data structure to be able to make a "backup" to a file and restore it later.

For now I've managed to export the entire structure as "DataValue", nothing too difficult:
<pre><code>async function read(session) { \n
    return new DataValue(await session.read(
    {
        nodeId: '...',
    }
));</code></pre>
To  save it i use the "fs" library.
If I don't interact with this variable and try to rewrite it directly on the server, everything works correctly.
The problems start when I try to save it to a file and then re-import it as DataValue.

Basically I can't get the "DataValue" object again, do you have any advice?

GabrieleFerronato, 2024-12-13:
> if you store it as json: https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-data-value/source/datavalue.ts#L241 it should be able to pass it in the DataValue-Constructor!?

I tried but if I do something like:
<pre><code>async function read(session) {
 const result = new DataValue(await session.read({
 nodeId: 'ns=3;s="Recipe_Controller01_D51_DataBaseAux"."THIS"[1]',
 }));
 return result.toJSON()
}
));</code></pre>

and after:
<pre><code>var x = new DataValue(await read(session))
</code></pre>

I have problems when I have to recreate the "Variant" class
<pre><code>Uncaught Error: A variant with DataType.ExtensionObject must have an ExtensionObject value.</code></pre>

AndreasHeine, 2024-12-13:
ah ok your variant is of type ExtensionObject....

https://github.com/AndreasHeine/SampleServer-node-opcua/blob/2388cef58130eb66a89ab32db1bd32567de5662d/src/machines/machinetool/showcasemachinetool.ts#L128-L135

if i remember correctly the session has a construct method as well!

GabrieleFerronato, 2024-12-13:
ok, I don't understand what parameters it needs:
<pre><code>dataType: NodeId,
pojo: Record<string, unknown>
</code></pre>
I imagine it needs a node from which to retrieve the ExtensionObject structure and the Json from which to retrieve the data, so the result would be:
<pre><code>var ExtensionObjJSON = (await read(session)).value; //{dataType: "ExtensionObject", arrayType: "Scalar", value: {...}}

var nodeId = 'ns=3;s="Recipe_Controller01_D51_DataBaseAux"."THIS"[1]';
var ExtensionObj = await session.constructExtensionObject(nodeId, ExtensionObjJSON)
</code></pre>

I tried and it doesn't work so I'm doing something wrong...

AndreasHeine, 2024-12-13:
because you pass a "string" instead of a "NodeId" ...
you try to create a structured datatype which is not part of the OPC UA Core Spec. means the client need to ask the server which structure the client has to download in order to assemble the ExtensionObject of that type!

`
const nid = coerceNodeId('ns=3;s="Recipe_Controller01_D51_DataBaseAux"."THIS"[1]')
`

the pojo is the actual object e.g. for a "Range" Structure:
https://reference.opcfoundation.org/Core/Part8/v104/docs/5.6.2
it looks like:
```
        {
            low: 60.0,
            high: 100.0,
        }
```

GabrieleFerronato, 2024-12-13:
ok, I understand, thanks for the explanation, now I can generate the ExtensionObject correctly.

As said, the ExtensionObject is syntactically correct but it has not taken the values ​​from the json, am I forgetting a step or should it retrieve them?

<pre><code>var ExtensionObjJSON = (await read(session)).value.value //.value.value to keep only the ExtensionObject data
let ExtObj = await session.constructExtensionObject(coerceNodeId(<StructureNodeID>, ExtensionObjectJSON))</code></pre>

AndreasHeine, 2024-12-13:
[exObj1, exObj2]

GabrieleFerronato, 2024-12-13:
fair enough, that's it for me, thanks a lot for the support! Have a nice weekend!

closed 2024-12-13 by GabrieleFerronato