node-opcua/node-opcua #1408: 2.140.0 Connect Error OpenSecureChannelRequest

zN3utr4l, 2025-01-13:
**Current behavior**

Install 2.140.0 version and connect to opc Ua

**Describe the bug**

Connect to Opc Ua show this error:

"The connection may have been rejected by server,\nErr = (Transaction has timed out ( timeout = 15000 ms , request = OpenSecureChannelRequest))"
  
"Error: The connection may have been rejected by server,\nErr = (Transaction has timed out ( timeout = 15000 ms , request = OpenSecureChannelRequest))\n    at <anonymous> (G:\\t-5\\server\\node_modules\\node-opcua-client\\source\\private\\client_base_impl.ts:981:27)\n    at <anonymous> (G:\\t-5\\server\\node_modules\\node-opcua-client\\source\\private\\client_base_impl.ts:752:29)\n    at <anonymous> (G:\\t-5\\server\\node_modules\\node-opcua-secure-channel\\source\\client\\client_secure_channel_layer.ts:642:17)\n    at G:\\t-5\\server\\node_modules\\async\\dist\\async.js:3682:19\n    at wrapper (G:\\t-5\\server\\node_modules\\async\\dist\\async.js:272:20)\n    at replenish (G:\\t-5\\server\\node_modules\\async\\dist\\async.js:442:29)\n    at iterateeCallback (G:\\t-5\\server\\node_modules\\async\\dist\\async.js:431:21)\n    at G:\\t-5\\server\\node_modules\\async\\dist\\async.js:328:20\n    at G:\\t-5\\server\\node_modules\\async\\dist\\async.js:3680:17\n    at <anonymous> (G:\\t-5\\server\\node_modules\\node-opcua-secure-channel\\source\\client\\client_secure_channel_layer.ts:637:25)"

When the connection is ok (2.139.0), these messages are shown:
08:52:39.963Z :populate_data_type_manager_104:62     Cannot find dataType Definition ! with nodeId =ns=2;i=6525 2:ParameterResultDataType false
08:52:40.004Z :populate_data_type_manager_104:62     Cannot find dataType Definition ! with nodeId =ns=2;i=15888 2:TransferResultErrorDataType false
08:52:40.005Z :populate_data_type_manager_104:62     Cannot find dataType Definition ! with nodeId =ns=2;i=15889 2:TransferResultDataDataType false

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Start Connect

**Expected behavior**

Connect to Opc Ua (2.139.0 work)

**Gist**

```
const client = OPCUAClient.create({
        applicationName: 'Client',
        connectionStrategy: {
          initialDelay: 1000,
          maxRetry: 1
        },
        securityMode: MessageSecurityMode.None,
        securityPolicy: SecurityPolicy.None,
        endpointMustExist: false,
        requestedSessionTimeout: 60000,
        keepSessionAlive: true,
        keepAliveInterval: 5000,
      })

      const endpointUrl = `opc.tcp://${this.ip}:${this.port}`
      await client.connect(endpointUrl)
```

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: Windows / Linux

    - (x) Windows : version : 11
    - (x) Linux : version : raspberry 
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Codesys
    - Version: 3.5 sp-20 patch 4
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :

2.140.0

 **Node:**

  `node --version` = 22.13.0

erossignon, 2025-02-07:
Sorry, there is not enough to reproduce here; 

Can you test with node-opcua@2.140.0 and 
```
typescript const endpoint = "opc.tcp://opcuademo.sterfive.com:26543";
```

 ? Can you connect ?

erossignon, 2025-02-07:
gist: 
```typescript
process.env.NODEOPCUADEBUG="CLIENT{TRACE-RESPONSE-REQUEST}";
process.env.NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE="10000" ;

const { OPCUAClient, MessageSecurityMode,   SecurityPolicy} = require("node-opcua");

async function main() {

    const client = OPCUAClient.create({
        applicationName: 'Client',
        connectionStrategy: {
          initialDelay: 1000,
          maxRetry: 1
        },
        securityMode: MessageSecurityMode.None,
        securityPolicy: SecurityPolicy.None,
        endpointMustExist: false,
        requestedSessionTimeout: 60000,
        keepSessionAlive: true,
        keepAliveInterval: 5000,
})

// const endpointUrl = `opc.tcp://${this.ip}:${this.port}`
const endpointUrl = `opc.tcp://opcuademo.sterfive.com:26543`;
await client.connect(endpointUrl);

const session = await client.createSession();   
await new Promise((resolve) => setTimeout(resolve, 100000));
await client.disconnect();
console.log("Done");
}
main();
```

zN3utr4l, 2025-02-10:
```console
07:40:36.857Z :verify                        :68    [NODE-OPCUA-W06] The certificate subjectAltName does not match the client applicationUri
 ...                                                Please regenerate a specific certificate that matches your client applicationUri
 ...                                                certificate subjectAltName  = urn:DESKTOP-P33D1A:AAAAA
 ...                                                client applicationUri  = urn:DESKTOP-P33D1A:Client
 ...                                                certificateFile         = C:\Users\beeeh\AppData\Roaming\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem

Done
```

It seems so, these are the logs shown in the console, the execution reaches "Done".

By updating to version 1.4.4 and using my initialization the error is no longer shown but it still doesn't seem to work, it doesn't accept writes and the logs that made me understand that everything was fine are not shown.

> When the connection is ok (2.139.0), these messages are shown:
>
> ```
> 08:52:39.963Z :populate_data_type_manager_104:62     Cannot find dataType Definition ! with nodeId =ns=2;i=6525 [...]

The subscription, session is created but nothing seems to work.

> Sorry, there is not enough to reproduce here;

This is the complete OpcUa Client service:

```typescript
import type { IDataNamesOpcUa, IDataNamesOpcUaObj, IDataOpcUa, IDate, ItemNames } from '@shared/opcUa.js'
import _ from 'lodash'
import {
  AttributeIds,
  type ClientSession,
  type ClientSubscription,
  DataType,
  MessageSecurityMode,
  type NodeId,
  OPCUAClient,
  type ReadValueIdOptions,
  SecurityPolicy,
  TimestampsToReturn,
  VariantArrayType,
  type WriteValueOptions,
  coerceNodeId
} from 'node-opcua'
import getLogger from './logger.js'
import { TypedEventEmitter } from './type-event-emitter.js'
const logger = getLogger(import.meta.filename)

interface IBrowseItem {
  name: keyof typeof ItemNames
  nodeId: string
  referenceTypeId: NodeId
  type?: DataType
  arrayType?: VariantArrayType
}

type Writeable<T> = { -readonly [P in keyof T]-?: T[P] }

interface OpcUAEventTypes {
  'opc_data_change': [arg1: Writeable<IDataOpcUa>],
  'opcua_status': [arg1: boolean]
}

class OPCUAClientService {
  public events: TypedEventEmitter<OpcUAEventTypes> = new TypedEventEmitter<OpcUAEventTypes>()
  private port = process.env.OPCUA_PORT ? parseInt(process.env.OPCUA_PORT) : 4840
  private ip = process.env.PLC_HOST ?? '127.0.0.1'
  private plcPath = 'ns=4;s=|var|CODESYS Control for Raspberry Pi.Application.'
  private countersPath = `${this.plcPath}GVL`
  private configVariablesPath = `${this.plcPath}GVL_Conf`
  private invertersPath = `${this.plcPath}GVL_Inverter`
[... 187 lines cut ...]
                dataType: item.type,
                value: newValue,
                arrayType: item.arrayType
              }
            }
          }
        }
        return null
      }))

      const filteredItems = itemsToWrite.filter(Boolean)
      if (filteredItems.length) {
        const result = await this.session!.write(filteredItems as WriteValueOptions[])
        const total = result.filter(x => x.isGood()).length
        logger.info(`Total Success/Passed ${total}/${items.length}`)
      }

    } catch (error) {
      logger.error(`Error writing range of variables into OPC UA server: ${error}`)
    }
  }
}

export const opcUaClientService = new OPCUAClientService()
```

erossignon, 2025-02-11:
I see several potential issues with your current implementation:

### No Need to Interfere with the Built-in Reconnection Mechanism

- One should rely on the OPCUAClient's ability to automatically handle reconnection behind the scenes.
- When the client.on("connection_lost") event is raised, you should treat this event as informational only. It indicates that the client is starting the automatic reconnection process. For example, you could set a flag to prevent reads, writes, or calls while the client is reconnecting automatically, as those actions would likely trigger exceptions.
- In your implementation, you're attempting to close the session and create a new client, but the previous client is not disposed of properly.
- My recommendation is not to interfere with the built-in reconnection mechanism and let the OPCUAClient restore the session automatically. Things should return to normal once you receive the client.on('connection_reestablished') event.

###  After the Session is Disposed, the Client Should Be Disconnected
- Don’t forget to explicitly disconnect the client after closing the session; otherwise, this could lead to unattended resource consumption on the server side.
-There’s a pattern that allows you to properly handle the lifetime of a connection. Instead of explicitly managing client.connection, client.createSession, session.close, client.disconnect, you can use this pattern:

```javascript
const client = OPCUAClient.create({});
client.withSubscriptionAsync(endpoint, subscriptionParameter, async (session, subscription) => {
    await session.read(...);
    const monitorItems = await subscription.monitoredItem(...);
    await waitForProgramTerminationRequest();
});
- The withSubscriptionAsync method automatically creates the connection, a session, and one subscription, then calls the async (session, subscription) method you provide. Once that method returns, withSubscriptionAsync will terminate the subscription, close the session, and properly disconnect the client.
Just make sure you have a method that waits for a signal to indicate that the OPC UA connection should be terminated. For example:

```javascript
await waitForProgramTerminationRequest();
```
or alternatively:
```javascript
await new Promise(resolve => process.on("SIGINT", resolve));
```

- This will wait for the user to press CTRL+C or any other appropriate termination signal from your main app.

### Dealing with a Large Number of Monitored Items in a Single Subscription
You could take advantage of [@sterfive/optimised-client](https://www.sterfive.com/product/premium-packages/), which facilitates the creation of a large number of monitored items by splitting requests behind the scenes in an optimized manner.

### Parallelism Issues with getWriteValue
- It seems that an await is missing in this.session!.constructExtensionObject(dataType, value);.

- The `constructExtensionObject` method fetches the Extension Object definition from the server the first time it is used by calling `extractNamespaceDataType`, which triggers several reads and browses behind the scenes. 
- You cannot perform constructExtensionObject for too many objects in parallel (as you are doing with await Promise.all(items.map(...))), as this will cause multiple fetches of the Extension Object definition simultaneously.

- One strategy is to preload the Extension Object map right after the connection is established, before any reads, writes, calls, or monitoring begins, by calling `await session.extractNamespaceDataType()` just after you get the session. This will ensure that everything is ready for constructExtensionObject to operate immediately|

You can gain more tips and expert advice by subscribing to the [NodeOPCUA Support membership](https://support.sterfive.com)

zN3utr4l, 2025-02-11:
Thanks for the advice, I tried to implement them, except `@sterfive/optimized-client`, for which I should purchase support, but that's not in my company's plans at the moment.

Despite the changes, **version 2.144.0 still doesn't work**, execution seems to stop at the `await this.session.extractNamespaceDataType()` line and no longer proceeds, but it doesn't fail with an error, I left it like that for 5 minutes and it didn't throw anything.

```typescript 
import type { IDataNamesOpcUa, IDataNamesOpcUaObj, IDataOpcUa, IDate } from '@shared/opcUa.js'
import _ from 'lodash'
import {
  AttributeIds,
  type ClientSession,
  type ClientSubscription,
  DataType,
  type NodeId,
  OPCUAClient,
  type ReadValueIdOptions,
  TimestampsToReturn,
  VariantArrayType,
  type WriteValueOptions,
  coerceNodeId
} from 'node-opcua'
import getLogger from './logger.js'
import { TypedEventEmitter } from './type-event-emitter.js'
const logger = getLogger(import.meta.filename)

interface IBrowseItem {
  name: IDataNamesOpcUa
  nodeId: string
  referenceTypeId: NodeId
  type?: DataType
  arrayType?: VariantArrayType
}

type Writeable<T> = { -readonly [P in keyof T]-?: T[P] }

interface OpcUAEventTypes {
  'opc_data_change': [arg1: Writeable<IDataOpcUa>],
  'opcua_status': [arg1: boolean]
}

class OPCUAClientService {
  public events: TypedEventEmitter<OpcUAEventTypes> = new TypedEventEmitter<OpcUAEventTypes>()
  private port = process.env.OPCUA_PORT ? parseInt(process.env.OPCUA_PORT) : 4840
  private ip = process.env.PLC_HOST ?? '127.0.0.1'
  private plcPath = 'ns=4;s=|var|CODESYS Control for Raspberry Pi.Application.'
  private countersPath = `${this.plcPath}GVL`
  private configVariablesPath = `${this.plcPath}GVL_Conf`
  private invertersPath = `${this.plcPath}GVL_Inverter`
  private session: ClientSession | null = null
  private browseItems: IBrowseItem[] = []
  private items: Writeable<IDataNamesOpcUaObj> = {} as IDataNamesOpcUaObj
  private isConnected = false
  private nodeIdReadable: ReadValueIdOptions[] = []
  private publishingInterval = 200
  private chunkSize = 100
  public getPlcVersion = () => `${this.items.Vers1_PLC}.${this.items.Vers2_PLC}.0`

  constructor() {
    try {
      const client = OPCUAClient.create({
        applicationName: 'Client',
        keepSessionAlive: true,
        keepAliveInterval: 5000
      })

      client.on('connection_lost', () => {
        logger.info('OPC UA connection lost')
        this.isConnected = false
        this.events.emit('opcua_status', this.isConnected)
      })

[... 121 lines cut ...]
                dataType: item.type,
                value: await this.getWriteValue(item, variable.value),
                arrayType: item.arrayType
              }
            }
          }
        }
        return null
      }))

      const filteredItems = itemsToWrite.filter(Boolean)
      if (filteredItems.length) {
        const result = await this.session!.write(filteredItems as WriteValueOptions[])
        const total = result.filter(x => x.isGood()).length
        logger.info(`Total Success/Passed ${total}/${items.length}`)
      }

    } catch (error) {
      logger.error(`Error writing range of variables into OPC UA server: ${error}`)
    }
  }
}

export const opcUaClientService = new OPCUAClientService()
```

erossignon, 2025-02-11:
So, you may have found an issue with extractNamespaceDataType  that can be exhibit with the particualr instance of the OPCUA Server you are using. 

let's narrow it down:

```typescript 
process.env.NODEOPCUADEBUG="CLIENT{TRACE-RESPONSE-REQUEST}";
process.env.NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE="10000" ;

const endpointUrl = `opc.tcp://<yourmachine>:<port>`;

const { OPCUAClient, MessageSecurityMode,   SecurityPolicy} = require("node-opcua");

async function main() {

  const client = OPCUAClient.create({});

  await client.withSessionAsync(endpointUrl, async (session) =>{
    console.log("calling extractNamespaceDataType ...");
    await session.extractNamespaceDataType();
    console.log("extractNamespaceDataType done...");
  });
  console.log("bye");
}
main();
```
* what is the outcome of the above script ?

* you can send us a mail at contact@sterfive.com so we can arrange a private debug session with live access to your OPCUA server for deeper investigation.

zN3utr4l, 2025-02-11:
stuck here for 2 minutes

```
13:18:42.944Z :verify                        :68    [NODE-OPCUA-W06] The certificate subjectAltName does not match the client applicationUri
 ...                                                Please regenerate a specific certificate that matches your client applicationUri
 ...                                                certificate subjectAltName  = urn:DESKTOP-AAAAAAAA:Client
 ...                                                client applicationUri  = urn:DESKTOP-AAAAAAAA:NodeOPCUA-Client
 ...                                                certificateFile         = C:\AppData\Roaming\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem
 ...
calling extractNamespaceDataType ...
```

erossignon, 2025-02-11:
OK, the process.env varaible should trigger the extensive log and we don't see it.
It would be ideal for me to be able to remote connect to your server deeper investigatinon
please contact us at  contact@sterfive.com

zN3utr4l, 2025-02-12:
I'm sorry but since it's an OpcUA server that isn't owned by me, I'm not authorized to let you connect.

If I can help you in any other way please let me know.

erossignon, 2025-02-12:
@zN3utr4l, the issue can be reproduced with any CodeSysServer will processing the Codessys dataType, on my side to.

zN3utr4l, 2025-02-13:
Thank you so much for testing with you too, I'm waiting for the new version.

erossignon, 2025-02-13:
@zN3utr4l, Please now check with [node-opcua@2.148.0] see release notes : https://github.com/node-opcua/node-opcua/releases/tag/v2.148.0.

Problem should be fixed.

zN3utr4l, 2025-02-14:
Problem solved, thank you very much

erossignon, 2025-02-19:
Closing as resolved.