node-opcua/node-opcua #1409: Does FileTranser feature support bind to a new file path

xky0007, 2025-01-21:
Support I already use 
```
installFileType(node, {
    filename: path
});
```

Can I also add a method to path a file path as input argument to update this node to bind to new path?

Thanks.

erossignon, 2025-01-27:
Yes, you can add a method to path a file path as input argument  to bind to new path.

Two stategy: 

* delete the existing UAFile node and recreate an other one, pointing to a different file.
* or (unsupported API) modify the filename using:
     ```
     (node as UAFileEx).$fileData.filename = "newfile";
     ```

You may want to consider our professional node-opcua modules @sterfive/file-transfer-server which provide additional functionality and support of [UAFileDirectory](https://support.sterfive.com)

xky0007, 2025-02-05:
@erossignon Thanks.
If I use `(node as UAFileEx).$fileData.filename = "newfile";`, will the old file handler closed? Will this automatically call `installFile` and do some method binding?

Thanks.

erossignon, 2025-02-05:
@xky0007:

> will the old file handler closed?   

probably not. you'll have to make sure that no client have opened the file and left it open while you are changing the file. Otherwise, it should be no harm.

> Will this automatically call installFile

InstallFile need to be called only once. It adds the file management behavior to the corresponding OPCUA node. 

You can still add support for the feature you need your self and provide a PR to enhance the library. Alternatively, contact us at contact@sterfive.com for paid consulting services if you want the maintainer to help you with this.

closed 2025-03-09 by erossignon