node-opcua/node-opcua #1410: handling sudden disconnection of server

faby291, 2025-01-22:
hello!

I'm  developing an app that allows me to import and export some data structure

But right now I'm facing some trouble in order to handle the sudden disconnection of the server es:(cable disconnection)

here is some part of my code:

 ```
 async function main() {
    try {
      // function to create client and session
      let OPCUA = await createSessionConn();
      const session = OPCUA[0];
      const client = OPCUA[1];
  
      //I find the node I want to check
      let activeControllers = await getActiveController(session);
  
      setControllerFlags(activeControllers);
  
      const rdy = getActiveControllerRdy();
  
      let y = 0;
      let id 
      let nodeData;

      // recorsive function for handling timer
      async function startTimer() {
        let timer = setInterval(async () => {
          y++;
          for (let i = 0; i < rdy.length; i++) {
           
            while (true) {
              try {
                  nodeData = await session.read({
                      nodeId: rdy[i],
                  });
                  if (i %2 == 0){
                    id = await session.read({
                      nodeId: `ns=3;s=\"Data1...".\"ExtID\"`,
                    });
                  }else{
                    id = await session.read({
                      nodeId: `ns=3;s=\"Data2...".\"ExtID\"`,
                    });
                  }
                  
                  break; 
              } catch (error) {
                  console.log(
                      `Failed to read node ${rdy[i]}, retrying...`
                  );
                  await new Promise((resolve) =>
                      setTimeout(resolve, 5000)
                  ); // Wait for 5 seconds before retrying
              }
            }
  
            if (nodeData.value.value && id.value.value === 1) {
                clearInterval(timer);
                await handleOperationChange(i, session, client);
                startTimer();
            }
          }
          console.log(y);
        }, 2000);
      }
  
      // Avvio del timer
      await startTimer();
    } catch (error) {
        console.error(error);
    }
  }
```

this is rdy: 
```
[
  'ns=3;s="node1..."."Rdy"',
  'ns=3;s="node2..."."Rdy"',
  'ns=3;s="node3..."."Rdy"',
  'ns=3;s="node4..."."Rdy"'
]
```

to handle disconnection I have:
```
 await new Promise((resolve) =>
                      setTimeout(resolve, 5000)
                  ); // Wait for 5 seconds before retrying
```

this work for me only if the disconnection is below 20 to 30 seconds

but if the disconnection is greather than 30 s, and I'm still cycling in order to see if any  of the provided node has changed state, than I'm unable to reconnect because i have to many read request and server can't keep up.

how can I fix it?

Thanks in advance

erossignon, 2025-01-27:
You need to check the session.isReconnecting flag  and not process the read if it is not set to true.

erossignon, 2025-01-27:
Rather than doing polling ( setTimer + cyclic read) please consider using subscriptions and monitoreed items. In this case, reconnection will be handled automatically by the server/client. 
It will also provide a optimum recovery of lost values, as this feature is built-in OPCUA subscription mecanism.

faby291, 2025-01-27:
Thanks, I'll try with session.isReconnecting, (that i didn't saw) and I'll let you know if it works.

For the subscription I can't use it, because I don't know how may rdy I will have to check, (it depends on the server opcua I'm connected), and what operation it does bring with the node itself

AndreasHeine, 2025-01-27:
> I don't get to work session.isReconnecting, So whenever I unplug the cable and check, its value won't change from false. Maybe I'm using it wrong?
> 
> I just added a check
> 
> ```
> if (session.isReconnecting){
>   console.log(`working`)
> }
> ```
> 
> before reading with session

you might look at which events get emitted to figgure out whats happening:

Client-Class:
```ts
        this.client.on("reconnection_attempt_has_failed", (err: Error, message: string) => {
            logger.error(`OPC UA Client: reconnect attemp has failed! err='${err}' message='${message}'`)
        })
        this.client.on("abort", () => {
            logger.error(`OPC UA Client: abort!`)
        })
        this.client.on("close", () => {
            logger.error(`OPC UA Client: close!`)
        })
        this.client.on("connection_failed", (err: Error) => {
            logger.error(`OPC UA Client: connection has failed! err='${err}'`)
        })
        this.client.on("connection_lost", () => {
            logger.warn(`OPC UA Client: connection lost!`)
        })
        this.client.on("connection_reestablished", () => {
            logger.log(`OPC UA Client: connection reestablished!`)
        })
        this.client.on("lifetime_75", (token: ChannelSecurityToken) => {
            logger.log(`OPC UA Client: securechannel token lifetime @ 75%! tokenId='${token.tokenId}'`)
        })
        this.client.on("receive_chunk", () => {
            // too noisy
        })
        this.client.on("receive_response", (response: Response) => {
            // too noisy
            // logger.log(`OPC UA Client: response='${response}'`)
        })
        this.client.on("security_token_renewed", () => {
            logger.log(`OPC UA Client: security token renewed!`)
        })
        this.client.on("send_chunk", () => {
            // too noisy
        })
        this.client.on("send_request", (request: Request) => {
            // too noisy
            // console.log(`OPC UA Client: request='${request}'`)
        })
        this.client.on("start_reconnection", () => {
            logger.log(`OPC UA Client: start reconnection!`)
        })
        this.client.on("timed_out_request", (request: Request) => {
            logger.warn(`OPC UA Client: request timed out! request='${request}'`)
        })
```

Session-Class:
```ts
        this.session.on("session_closed", (statusCode: StatusCode) => {

        })
        this.session.on("keepalive", (lastKnownServerState: ServerState) => {

        })
        this.session.on("session_restored", () => {

        })
        this.session.on("keepalive_failure", (state: any) => {

        })
```

faby291, 2025-01-27:
Thanks, a lot! It works! 

After cleaning and reorganising some code I made it work, now I able to handle the event of disconnection

if someone want to see how I did here is the code:

```

async function main() {
  try {
    // Create session and client
    let OPCUA = await createSessionConn();
    let session = OPCUA[0];
    let client = OPCUA[1];

     // Ensure the client and session exist before proceeding
     if (!client || !session) {
      throw new Error("Client or session not initialized properly.");
  }

  client.on("connection_reestablished", () => {
      console.log("OPC UA Client: connection reestablished!");
      startTimer();
  });

  client.on("lifetime_75", (token) => {
    console.log(`OPC UA Client: securechannel token lifetime @ 75%! tokenId='${token.tokenId}'`)
  })

  session.on("session_restored", () => {
      console.log("Session restored!");
  });

  console.log("Event listeners attached successfully!");

    //I find the node I want to check
    let activeControllers = await getActiveController(session);

    setControllerFlags(activeControllers);

    const rdy = getActiveControllerRdy();

    console.log(rdy);

    let y = 0;
    let id 
    // recursive function
    async function startTimer() {
      while (true) {
        y++;
        for (let i = 0; i < rdy.length; i++) {
          let nodeData;
          while (true) {
            try {
              // Timeout of 2 s
              nodeData = await Promise.race([
                session.read({
                  nodeId: rdy[i],
                }),
                new Promise((_, reject) =>
                  setTimeout(() => reject(new Error("Read timed out")), 2000)
                ),
              ]);
              if (i % 2 == 0) {
                id = await Promise.race([
                  session.read({
                    nodeId: `ns=3;s=\"Recipe...".\"ExtID\"`,
                  }),
                  new Promise((_, reject) =>
                    setTimeout(() => reject(new Error("Read timed out")), 2000)
                  ),
                ]);
              } else {
                id = await Promise.race([
                  session.read({
                    nodeId: `ns=3;s=\"Recipe...".\"ExtID\"`,
                  }),
                  new Promise((_, reject) =>
                    setTimeout(() => reject(new Error("Read timed out")), 2000)
                  ),
                ]);
              }
    
              break; 
            } catch (error) {
              if (error.message === "Read timed out") {
                console.log(`Timeout: Read operation took too long for node ${rdy[i]}`);
                return; 
              }else{
                return;
              }
            }
          }
    
          if (nodeData && nodeData.value.value && id.value.value === 1) {
            await handleOperationChange(i, session, client);
            return startTimer(); 
          }
        }
        console.log(y);
        await new Promise((resolve) => setTimeout(resolve, 2000)); 
      }
    }
    

    // Avvio del timer
    await startTimer();

    console.log('I am out');
    client.on("connection_lost", () => {
      console.warn("OPC UA Client: connection lost!");
    });
    
  } catch (error) {
      console.error(error);
  }
}
```

closed 2025-01-27 by faby291

erossignon, 2025-01-28:
A smarter solution based on Subscription might look like:

```typescript
import { OPCUAClient, DataValue, AttributeIds, IBasicSession } from "node-opcua-client";
async function handleOperationChange(session: IBasicSession , dataValue: DataValue) {
}
async function main() {

   const client = OPCUAClient.create({});
   await client.withSubscriptionAsync(endpointUrl, subscriptionParameters, async (session, subscription) => {
        const nodeId = "ns=3;s=\"Recipe...".\"ExtID\"";
        const monitorItem = await subscription.monitor({ nodeId: attributeId: AttributeIds.Value }, { samplingInterval: 1000});
        monitoredItem.on("changed", (dataValue)=> {
             handleOperationChange(session, dataValue)
            .then(()=>{})
            .catch((err)=> console.log(err.message));
        });
        // wait for user to press CTRL+C to terminate
        await Promise((resolve)=>process.on("SIGINT",resolve));
   });
}
```