node-opcua/node-opcua #1413: Creating Custom ExtensionObject

cihanyrtsvr, 2025-02-01:
**Title: Error When Creating a Custom ExtensionObject with Node-OPCUA**

**Description:**

Hello,

I am encountering an issue while trying to create a custom extension object with Node-OPCUA. Specifically, I am getting the following error:

```prompt
Error: Cannot find StructureType constructor for dataType ns=1;i=1000 at DataTypeFactory.findStructureInfoForDataType at ExtraDataTypeManager.getExtensionObjectConstructorFromDataType at AddressSpace.getExtensionObjectConstructor at AddressSpace.constructExtensionObject at main (file:///path/to/your/file)
```

Here is the code I am using:

```javascript
import { OPCUAServer, DataType, Variant } from "node-opcua";

async function main() {
  // Create the OPC UA server instance
  const server = new OPCUAServer({
    port: 4840,
    resourcePath: "/UA/MyCustomOPCUAServer",
    buildInfo: {
      productName: "MyCustomOPCUAServer",
      buildNumber: "1",
      buildDate: new Date()
    }
  });

  await server.initialize();
  console.log("Server initialized");

  const addressSpace = server.engine.addressSpace;
  const namespace = addressSpace.getOwnNamespace();

  // Step 1: Create a Custom Data Type with Double and Boolean fields
  const myCustomDataType = namespace.createDataType({
    browseName: "MyCustomDataType",
    isAbstract: false,
    description: "A custom data type with Double and Boolean fields",
    subtypeOf: "Structure",
    fields: [
      { name: "temperature", fieldType: "Double" },
      { name: "isActive", fieldType: "Boolean" }
    ]
  });

  console.log("Registered custom data type 'MyCustomDataType'");

  // Step 2: Construct an ExtensionObject of the Custom Data Type with values
  const myDataInstance = addressSpace.constructExtensionObject(myCustomDataType, {
    temperature: 25.5,
    isActive: true
  });

  console.log("Custom data type instance created:", myDataInstance);

  // Step 3: Add a custom object to the address space
  const myObject = namespace.addObject({
    browseName: "MyCustomObject",
    organizedBy: addressSpace.rootFolder.objects
  });

  // Step 4: Add a variable to hold the custom data type instance
  namespace.addVariable({
    browseName: "MyCustomDataVariable",
    dataType: "MyCustomDataType",
    value: {
      dataType: DataType.ExtensionObject,
      value: myDataInstance
    },
    organizedBy: myObject
  });

  console.log("Custom variable with custom data type added");

  // Step 5: Start the server
  await server.start();
  console.log("Server is now listening at:", server.endpoints[0].endpointDescriptions()[0].endpointUrl);

  // Handle shutdown gracefully
  process.on("SIGINT", async () => {
    console.log("Shutting down server...");
    await server.shutdown(1000);
    console.log("Server shut down");
    process.exit(0);
  });
}

main().catch((err) => {
  console.error("Error:", err);
});
```

**Problem:**

* I am trying to create a custom extension object that contains fields like Boolean, Double, Integer, and String.
* However, I keep encountering the error above when I attempt to create the ExtensionObject.
* Specifically, it seems like the StructureType constructor for my custom data type cannot be found.

**Request for Help:**

Could someone guide me on how to correctly create and instantiate a custom *ExtensionObject* with the specified fields? Additionally, any advice on how to avoid or resolve the `StructureType` constructor error would be greatly appreciated.

Thank you!

erossignon, 2025-02-04:
createDataType is not sufficient as you would also need to create the nodeId for serialiazation (encoding nodeIds)

This is where the OPCA Modelder comes to the rescue.

You can use the [OPCUA Modeler developed by Sterfive (the author of NodeOPCUA)](https://opcua-modeler.doc.sterfive.com/)  to generate a companion specification that contains your new types. 

The syntax of the model in yaml is very easy: 

```yaml
# model.yaml
# yaml-language-server: $schema=https://support.sterfive.com/nodeset2.schema.json
namespaceUri: http://sterfive.com/UA/MyModel/
version: 1.0.0
publicationDate: 2019-01-01T12:00:00Z

dataTypes:
  - browseName: MyCustomDataType
    isAbstract: false
    description: A custom data type with Double and Boolean fields
    fields:
      - name:  Temperature
        dataType: ua:Double
      - name: IsActive
        dataType: ua:Boolean
```

Then use the command:

```
opcua-modeler geneate -i model.yaml
```

This will generate a node set xml `model.NdoeSet2.xml` file that you can load to your server

```xml
<?xml version="1.0"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:ns1="http://sterfive.com/UA/MyModel/Type.xsd">
    <NamespaceUris>
        <Uri>http://sterfive.com/UA/MyModel/</Uri>
    </NamespaceUris>
    <Models>
        <Model ModelUri="http://sterfive.com/UA/MyModel/" Version="1.0.0" PublicationDate="2019-01-01T12:00:00.000Z">
            <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.05.03" PublicationDate="2023-12-15T00:00:00.000Z"/>
        </Model>
    </Models>
    <Aliases>
        <Alias Alias="HasEncoding">i=38</Alias>
        <Alias Alias="HasSaubtype">i=45</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
    </Aliases>
<!--ReferenceTypes-->
<!--DataTypes-->
    <UADataType NodeId="ns=1;i=1000" BrowseName="1:MyCustomDataType">
        <DisplayName>MyCustomDataType</DisplayName>
        <Description>A custom data type with Double and Boolean fields</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=1001</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=1002</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=1003</Reference>
        </References>
        <Definition Name="1:MyCustomDataType">
            <Field Name="Temperature" DataType="i=11"/>
            <Field Name="IsActive" DataType="i=1"/>
        </Definition>
    </UADataType>
    <UAObject NodeId="ns=1;i=1001" BrowseName="Default Binary">
        <DisplayName>Default Binary</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
        </References>
    </UAObject>
    <UAObject NodeId="ns=1;i=1002" BrowseName="Default XML">
        <DisplayName>Default XML</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
        </References>
    </UAObject>
    <UAObject NodeId="ns=1;i=1003" BrowseName="Default JSON">
        <DisplayName>Default JSON</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
        </References>
    </UAObject>
<!--ObjectTypes-->
<!--VariableTypes-->
<!--Other Nodes-->
</UANodeSet>
```

Then you can modify your server code to load the companion NodeSet and adapt the code this way:

```typescript
import { OPCUAServer, DataType, nodesets } from "node-opcua";

async function main() {
    // Create the OPC UA server instance
    const server = new OPCUAServer({
        port: 4840,
        resourcePath: "/UA/MyCustomOPCUAServer",
        buildInfo: {
            productName: "MyCustomOPCUAServer",
            buildNumber: "1",
            buildDate: new Date()
        },
        nodeset_filename: [
            nodesets.standard,
            "./model.Nodeset2.xml"
        ]
    });

    await server.initialize();
    console.log("Server initialized");

    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();

    const nsMyModel = addressSpace.getNamespaceIndex("http://sterfive.com/UA/MyModel/");

    const myCustomDataTypeNode = addressSpace.findDataType("MyCustomDataType", nsMyModel);
    if (!myCustomDataTypeNode) {
        throw new Error("Custom data type not found in the address space");
    }   
    const myCustomDataType = myCustomDataTypeNode.nodeId;

  

    // Step 2: Construct an ExtensionObject of the Custom Data Type with values
    const myDataInstance = addressSpace.constructExtensionObject(myCustomDataType, {
        Temperature: 25.5,
        IsActive: true
    });

    console.log("Custom data type instance created:", myDataInstance);

    // Step 3: Add a custom object to the address space
    const myObject = namespace.addObject({
        browseName: "MyCustomObject",
        organizedBy: addressSpace.rootFolder.objects
    });

    // Step 4: Add a variable to hold the custom data type instance
    namespace.addVariable({
        browseName: "MyCustomDataVariable",
        dataType: myCustomDataType,
        value: {
            dataType: DataType.ExtensionObject,
            value: myDataInstance
        },
        organizedBy: myObject
    });

    console.log("Custom variable with custom data type added");

    // Step 5: Start the server
    await server.start();
    console.log("Server is now listening at:", server.endpoints[0].endpointDescriptions()[0].endpointUrl);

    // Handle shutdown gracefully
    process.on("SIGINT", async () => {
        console.log("Shutting down server...");
        await server.shutdown(1000);
        console.log("Server shut down");
        process.exit(0);
    });
}

main().catch((err) => {
    console.error("Error:", err);
});

```

et voilà:

[image]

cihanyrtsvr, 2025-02-16:
Thanks so much for your help! I really appreciate you looking into this. As a student and someone just starting out with node-opcua, this was incredibly useful.

erossignon, 2025-02-19:
Closing as resolved.

closed 2025-02-19 by erossignon