node-opcua/node-opcua #1419: Can't raise event with multiple properties of same name

LaurenceGA, 2025-02-21:
**Current behavior**

I have a custom event type, which I'm creating like so:

```typescript
const namespace = addressSpace.getOwnNamespace();
const customEventType = namespace.addEventType({
  browseName: "ClashingSeverityEventType",
  subtypeOf: addressSpace.findObjectType(ObjectTypeIds.BaseEventType)!,
});

// Add custom properties to the event type that clashes with 0:Severity
namespace.addVariable({
  propertyOf: customEventType,
  browseName: "Severity",
  dataType: "UInt32",
  modellingRule: "Mandatory",
});
```

**Describe the bug**

When I raise an event like:

```typescript
const sourceNode = addressSpace.findNode("ns=0;i=84")
sourceNode.raiseEvent("1:ClashingSeverityEventType", {
  severity: {
    dataType: DataType.UInt32,
    value: 1,
  },
  ["1:severity"]: { // ??
    dataType: DataType.UInt32,
    value: 2,
  }
})
```

Event names are specified with the lowercase of the browse name, but since those are not fully qualified, that leads to clashes, meaning I cannot specify custom properties like this.

Trying to specify my extra property I get a log message like:

```
constructEventData:  cannot find property '1:severity' in [ eventId, eventType, sourceNode, sourceName, time, receiveTime, localTime, message, severity, conditionClassId, conditionClassName, conditionSubClassId, conditionSubClassName] when filling 1:ClashingSeverityEventType
```

As you can see, it only has one "severity", so it seems like I don't even have an option to choose which property. Or maybe I'm missing something?

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Create custom event type as above
2. Raise an event and try to specify your custom property
3. See that it is not an option

**Expected behavior**

Properties are namespaced, and when specifying properties, you should be able to say which namespace it's in so you can fully model an event.

**Screenshots**

N/A

**Gist**

N/A

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - (X) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

N/A

- **node-opcua version:** :

`2.143.0`

- **Node:**

  `node --version` =`v18.20.5`

AndreasHeine, 2025-02-21:
@LaurenceGA sometimes you need to install the extra properties:

`demoEvent.install_extra_properties();`

in order to use them!

https://github.com/AndreasHeine/SampleServer-node-opcua/blob/c404cb2c751a813f5d773b66e76fc73f44e19f4e/src/serveraddressspace/serveraddressspace.ts#L259C3-L259C12

LaurenceGA, 2025-02-21:
Interesting, it seems to still work for non clashing properties without having to install them, so I'm not sure exactly what that method actually does

erossignon, 2025-02-23:
node-opcua is not inhume to issue if browsename clashes at this time. 
It tries to makes things simple on the API side to avoid thinking about namespace index in all circunstancies. This comes with some limitations when browseanme clashes happen. 

A solution would be to create a raiseEventEx method that takes into account those rare cases where everything need to be specify with namespace explicly.

Let's wait and see if a sponsor raises the same request, so we can plan the associated effort.

erossignon, 2025-02-23:
Why not reuse existing Severity property in your case ?
Do you need to distinguish between UA standard Priority and an other one  ?

LaurenceGA, 2025-02-23:
Thanks for the confirmation.

> A solution would be to create a raiseEventEx method that takes into account those rare cases where everything need to be specify with namespace explicly.

Makes sense, or even extend the current one to accept `RaiseEventData | ExtendedRaiseEventData` or something

> Why not reuse existing Severity property in your case ?
Do you need to distinguish between UA standard Priority and an other one ?

It was only to provide a simple illustrative example, not a real use case.
I was writing some code to make some subscriptions, and wanted to construct various server test cases to make sure it handled them properly, and this is a possible edge case to handle. I don't have any data for how common this might be.