node-opcua/node-opcua #1421: Check if the value is a valid NodeId

Teo230, 2025-03-10:
-->

**Is your feature request related to a problem? Please describe.**
Yes, I'm developing a node-red opc package that is using this library.

**Describe the solution you'd like**
Create a new static method that returns a boolean if the value is a valid NodeId
(e.g. i=84 => true, i0=fef => false)

**Describe alternatives you've considered**
I'm currently using try-catch block in order to verify if the value is a valid NodeId.

**Additional context**

Add any other context or screenshots about the feature request here.

AndreasHeine, 2025-03-10:
you could use a regex!?
according to copilot the following could work:  
```^(ns=\d+;(?:i=\d+|s=[\w\-]+|g=[\da-fA-F\-]+|b=[\da-fA-F]*))$```

Teo230, 2025-03-10:
Good solution, my request was to integrate directly into the library in the NodeId class

erossignon, 2025-03-17:
```
function isValidNodeIdString(nodeIdString: string): boolean {
  try {
    const nodeId = coerceNodeId(nodeIdString);
    return true;
  } catch(err) {
    return false;
  }
}
```

closed 2025-03-17 by erossignon