node-opcua/node-opcua #1422: Connection issue between node-opcua client and C# opcua server

Nickiven, 2025-03-13:
Hi there,

Current I'm trying to use node-opcua client to connect with my C# opcua server. Both client and server run my **local** PC.

After connection established, everything goes smoothly. I also subscribe a monitor to watch variable changed event. But after 7min later, I get an issue as below:

[image]

Then I lost my connection.

[image]

Then I use my C# opcua client to repeat this problem. But connection is good, and no error raised.
My question is why my node-opcua client can't refresh server clock. I compare client code between node and C#. I can't figure out. Btw, I haven't config any certificate for server and client.

C# client code:
            var config = new ApplicationConfiguration()
            {
                ApplicationName = "OPCUAClient-C#",
                ApplicationType = ApplicationType.Client,
                SecurityConfiguration = new SecurityConfiguration
                {
                    ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "Directory",
                        StorePath = "Certificates/Client",
                        SubjectName = "OPCUAClient"
                    },
                    TrustedPeerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "Certificates/TrustedPeers"
                    },
                    TrustedIssuerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "Certificates/TrustedIssuers"
                    },
                    RejectedCertificateStore = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "Certificates/RejectedCertificates"
                    },
                    AutoAcceptUntrustedCertificates = true
                },
                TransportConfigurations = new TransportConfigurationCollection(),
                TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
                ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 },
                TraceConfiguration = new TraceConfiguration()
            };
            await config.Validate(ApplicationType.Client);

            var endpointUrl = "opc.tcp://******";
            var selectedEndpoint = CoreClientUtils.SelectEndpoint(endpointUrl, useSecurity: false, 15000);
            var endpointConfiguration = EndpointConfiguration.Create(config);
            var endpoint = new ConfiguredEndpoint(null, selectedEndpoint, endpointConfiguration);
            _session = await Session.Create(config, endpoint, false, "OPCUAClient", 60000, null, null);
node client code:
this.client = opcua.OPCUAClient.create({
        // applicationName: 'UVN OpcUa Client',
        endpointMustExist: false,
        keepSessionAlive: true,
        requestedSessionTimeout: 60000,
        // clientCertificateManager: certificateManager,
      })
      await this.client.connect(endpointUrl)
      this.session = await this.client.createSession()

I'm a new opcua player. Please help me, thanks.

OS: windows10
Node version: 20.11.1
Node-opcua version: ^2.149.0

erossignon, 2025-03-17:
Creating a C# server with OpcFoundation OpcUa module runing on a machine with 7 minutes discrepancy does't show the issue. 

There is not enough information on the Server Side, please be more specific about the C# Server -  mimial code for the server and the stack+version used ?

Nickiven, 2025-03-24:
> Creating a C# server with OpcFoundation OpcUa module runing on a machine with 7 minutes discrepancy does't show the issue.
> 
> There is not enough information on the Server Side, please be more specific about the C# Server - mimial code for the server and the stack+version used ?
> 
> In particular (as stated in the issue template)
> 
> * Description of the other OPCUA system I am trying to connect to:
>   
>   * Name: ______
>   * Version: ______
>   * Manufacturer/Software vendor: ______
>   * link to the manufacturer or product site: https://

Hi erossignon,

Very appreciate for your help. Please check my server code. First, I referenced a library. Details as below:

* Name: OPCFoundation.NetStandard.Opc.Ua
* Version: 1.5.375.443
* Manufacturer/Software vendor: OPCFoundation
* Link to the manufacturer or product site: [https://](https://github.com/OPCFoundation/UA-.NETStandard)

C# code:
```
_appInstance = new ApplicationInstance
{
    ConfigSectionName = "OpcUaServer"
};

var config = await _appInstance.LoadApplicationConfiguration(false);

if (!await _appInstance.CheckApplicationInstanceCertificate(false, 0))
{
    Console.WriteLine("CheckApplicationInstanceCertificate failed");
}

if (!config.SecurityConfiguration.AutoAcceptUntrustedCertificates)
{
    config.CertificateValidator.CertificateValidation += CertificateValidator_CertificateValidation;
}

await _appInstance.Start(_opcUaServer);
```
OpcUaServer
```
using Opc.Ua;
using Opc.Ua.Server;

namespace OPCUAServer
{
    public class OpcUaServer : StandardServer
    {
        private readonly INodeManagerFactory _nodeManagerFactory;

        public OpcUaServer(INodeManagerFactory nodeManagerFactory)
        {
            _nodeManagerFactory = nodeManagerFactory;
        }

        protected override MasterNodeManager CreateMasterNodeManager(IServerInternal server, ApplicationConfiguration configuration)
        {
            return new MasterNodeManager(server, configuration, null, new INodeManager[] { _nodeManagerFactory.Create(server, configuration) });
        }

        protected override void OnServerStarted(IServerInternal server)
        {
            base.OnServerStarted(server);
        }

        protected override void OnServerStopping()
[... 125 lines cut ...]
      </SamplingRateGroup>
      <SamplingRateGroup>
        <Start>1000</Start>
        <Increment>500</Increment>
        <Count>20</Count>
      </SamplingRateGroup>
    </AvailableSamplingRates>
    <MaxRegistrationInterval>30000</MaxRegistrationInterval>
    <NodeManagerSaveFile>OpcUaServer.nodes.xml</NodeManagerSaveFile>
    <DiscoveryUrls>
      <Url>opc.tcp://localhost:4840</Url>
    </DiscoveryUrls>
  </ServerConfiguration>

  <TraceConfiguration>
    <OutputFilePath>Logs\SampleOpcUaServer.log</OutputFilePath>
    <DeleteOnLoad>true</DeleteOnLoad>
    <TraceMasks>515</TraceMasks>
  </TraceConfiguration>

</ApplicationConfiguration>
```

I have not config any token. Why the client can't refresh token data...
I'm a new OpcUa player. Thanks for your help!