node-opcua/node-opcua #1426: Cannot find module '\node-opcua-crypto\dist\source\index_web.js' after packaging

francescozanotto, 2025-03-31:
I'm using node-opcua package in a nodejs program.

The program works fine, but I get the error linked in object after packaging using [pkg](https://www.npmjs.com/package/pkg) package.

I opened this issue in node-opcua-crypto but maybe this is the correct place to place it:
https://github.com/node-opcua/node-opcua-crypto/issues/66

erossignon, 2025-04-06:
Unfortunatly [`pkg`](https://github.com/vercel/pkg) is deprecated and doesn't support modern nodejs version. 

Various alternative:
  - use a bundler such as tsup , esbuild or webpack to convert your code to es6 prior to use pkg
  - use the SEA feature of nodejs 23 ( still active developpment)

LeMarlish, 2025-04-09:
I got it to work with [Esbuild](https://esbuild.github.io/getting-started/)!

First, you need to bundle your project together
```
esbuild index.js --bundle --outputfile=myapp.js --platform=node
```

Then run pkg as usual
```
pkg myapp.js <your params>
```

It'll generate myapp.exe.

If you prefer to compile using `pkg package.json` instead, this is the configuration I used:
```
"bin": "myapp.js",
  "pkg": {
    "scripts": "myapp.js",
    "assets": [
      "./nodesets/*"
    ],
    "targets": [
      "node*-win-x64"
    ],
    "outputPath": "bin"
  }
```

Note: If you use the premade nodesets from the module, I recommend using your own nodeset files or copy the ones you're using from **node_modules/node-opcua-nodesets/nodesets** into your project and add them to the bundle, as the app will say it couldn't find your nodesets at ./nodesets/file.xml otherwise, unless you make a folder called nodesets at the same level as the executable with the xml files you use inside.

closed 2025-04-10 by erossignon