node-opcua/node-opcua #1427: [Client] node-opcua client on linux / macOS does not reconnect if OPC server becomes unreachable

alex-ioma, 2025-04-04:
**Current behavior**  

When using a node-opcua client on a linux system to connect to an OPC UA server, the client does not handle reconnection and freezes:

```js
      const connectionStrategy = {
        initialDelay: 1000,
        maxDelay: 12 * 1000,
        maxRetry: -1 // Infinite retries
      };

      const options = {
        connectionStrategy,
        keepSessionAlive: true,
        reconnectOnFailure: true,
        keepAliveInterval: 500,
        operationTimeout: 500,
        endpointMustExist: false,
        keepPendingSessionsOnDisconnect: true 
      };

      const client: OPCUAClient = OPCUAClient.create(options);
      client.on('connected', async (retry, delay) => {
      const session: ClientSession = await this.createSession(plcUrl, client);
      })
```
- If the **OPC UA server** is stopped but the **OPC server remains reachable (pingable)**, the client correctly attempts to reconnect.
```
Client PLC Event: connection_lost opc.tcp://localhost:4840
Client PLC Event: start_reconnection opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
Client PLC Event: backoff opc.tcp://localhost:4840
...
```
- However, if the **entire OPC server is disconnected / become unreacheable**, the client closes and does not attempt to reconnect.  
```
14:46:03.953Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
14:46:19.457Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
14:46:34.963Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
14:46:50.468Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)
Client PLC Event: timed_out_request opc.tcp://192.168.88.50:4840
Client PLC Event: close opc.tcp://192.168.88.50:4840
14:46:58.431Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server   Connection has been closed by client , but this transaction cannot be honored
14:46:58.939Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:46:59.442Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:46:59.946Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:00.449Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:00.952Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:01.455Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:01.958Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:02.460Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:02.964Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:03.467Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
14:47:03.971Z :client_session_keepalive_manag:156    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
...
```

---

**Describe the bug**  
When the OPC UA server is disconnected, the node-opcua client does not attempt reconnection correctly on linux.

The same behavior on Windows is not present and the client **always** reconnects correctly.

---

**Step by step instruction to reproduce**  

1. Start a node-opcua client and connect it to an OPC UA server running on a PLC.  
2. Disconnected the OPC server (so it's completely unreachable).  
3. Observe that the client does not attempt reconnection.  
4. Compare this behavior to stopping only the OPC UA server process while keeping the server reacheable by ping - the client correctly attempts to reconnect indefinitely. 

---

**Expected behavior**  
The node-opcua client should **always** attempt to reconnect, even when the entire server is turned off or unreachable. When the server is powered back on, the client should successfully re-establish the connection without requiring a manual restart.

---

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (X) Linux : version : ______
    - (X) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: SIMATIC S7-1500
    - Manufacturer/Software vendor: Siemens

- **node-opcua version:** :

2.150.0

- **Node:**

  `node --version` = v23.7.0

erossignon, 2025-04-05:
you need to connect first before creating the session.

```typescript

  const client: OPCUAClient = OPCUAClient.create(options);

  await client.connect(plcUrl);

  const session = await client.createSession();

``̀`

Please refer to my book [NodeOPCUA by example](https://leanpub.com/node-opcuabyexample-edition2024) to get best practices and pattern to develop clients application.

alex-ioma, 2025-04-05:
Thank you @erossignon for the quick reply. `client.connect()` is already invoked before the session is created. It was just missing from the snippet above. 

I'm afraid the issue is there and, indeed, I try and troubleshoot the code for a week before hunting down the root cause and opening the issue.

Root cause: if the server/plc completely becomes unavailable by ping the library cannot recover after the server/plc becomes available again.

Proof/findings:
- When run on linux the library does not reconnect if goes offline. But it does reconnect if the plc just stops and restarts the OPC server but remains available by ping.
- The exact same code on Windows works with no problem in all scenarios (i.e. library always reconnects even after plc goes offline by png).
- Exact same code on linux always reconnects even after plc goes offline **if a proxy is added in front of the plc** - i.e. even if the plc goes offline, the "virutal IP" of the opc server is always pingable.

Can you please help us in solving this issue?

erossignon, 2025-04-06:
When cable is unplugged or ethernet card  is disabled nodejs might not detect connection lost easily.
[node-opcua@2.151.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.151.0) now forces a connection break when the keep alive hit a request timeout, forcing the reconnection Process to kick-off.

Please give it a carefull try and let me know if it is now working better.

alex-ioma, 2025-04-07:
@erossignon we have tested it today and everything is working as expected. Thank you very much for the support.

closed 2025-04-07 by alex-ioma