node-opcua/node-opcua #1429: node-opcua-client error instead of backoff when connecting to unreachable server with v2.151.

wrnotc, 2025-04-08:
**Current behavior**

When trying to connect to an unreachable / none existing OPC UA server in version v2.151.0 of node-opcus-client, the client does not backoff, but aborts instead.

**Describe the bug**

After switching from node-opcua-client v2.150.0 to v2.151.0 the client won't connect property with automatic retries. 
In v2.150.0 the server configured with e.g.
```mts
        const connectionStrategy: ConnectionStrategyOptions = {
            maxRetry: -1, // infinit retries
            initialDelay: 1 * 1000,
            maxDelay: 10 * 1000
        };
```
would automatically try to connect to a OPC UA server with infinit retries, but wit v2.151.0 the connection is aborted and an an error is thrown. 

Might have something to do with: [Issue 1427](https://github.com/node-opcua/node-opcua/issues/1427)

**Step by step instruction to reproduce**

Full example code:
```mts
// This file shall show the connection problems with the v2.151 of node-opcua-client in comparison to v2.150.0.

import {
    ConnectionStrategyOptions,
    OPCUAClient,
    OPCUAClientOptions,
} from "node-opcua-client";
import EventEmitter from "node:events";

/**
 * Function to log every emitted event of an event emitter.
 * @param emitter The event emitter to patch.
 * @param logName The name to log with.
 */
export function patchEmitter(emitter: EventEmitter) {
    var oldEmit = emitter.emit;

    emitter.emit = function (eventName: string, ...eventArgs: any[]): boolean {
        // Log the event name and arguments
        const now = new Date();
        console.log(new Date().toISOString(), "Event emitted:", eventName);

        // Call the original emit method
        return oldEmit.apply(emitter, [eventName, ...eventArgs]);
    };
}

const endpointUrl = "opc.tcp://10.20.30.40:4840"; // arbitrary IP, no opc ua server shall exist at this address

async function main() {
    try {

        // setup  connect with infinit retries
        const connectionStrategy: ConnectionStrategyOptions = {
            maxRetry: -1, // infinit retries
            initialDelay: 1 * 1000,
            maxDelay: 10 * 1000
        };

        const connectOptions: OPCUAClientOptions = {
            connectionStrategy: connectionStrategy,
            endpointMustExist: false,
            keepSessionAlive: true,
            tokenRenewalInterval: 5 * 1000,
            transportTimeout: 10 * 1000, // send backoff every 10 seconds
        };

        const client = OPCUAClient.create(connectOptions);
        patchEmitter(client); // log every event emitted by client

        await client.connect(endpointUrl);

    } catch (err) {
        console.log('Error:', err);
    }
}

main();
```

**running the code with v2.150.0**
When running the code with v2.150.0 the output looks like this:
```sh
2025-04-08T06:44:42.969Z Event emitted: secure_channel_created
2025-04-08T06:44:52.975Z Event emitted: backoff
2025-04-08T06:45:03.980Z Event emitted: backoff
2025-04-08T06:45:15.986Z Event emitted: backoff
2025-04-08T06:45:29.990Z Event emitted: backoff
2025-04-08T06:45:47.992Z Event emitted: backoff
2025-04-08T06:46:07.994Z Event emitted: backoff
2025-04-08T06:46:27.995Z Event emitted: backoff
2025-04-08T06:46:47.997Z Event emitted: backoff
2025-04-08T06:47:07.998Z Event emitted: backoff
```

**running the code with v2.151.0**
When running the code with v2.150.1 the output looks like this:
```
2025-04-08T06:48:24.476Z Event emitted: secure_channel_created
2025-04-08T06:48:34.481Z Event emitted: abort
2025-04-08T06:48:34.482Z Event emitted: connection_failed
Error: Error: The connection may have been rejected by server,
Err = (ECONNRESET)
    at <anonymous> (/home/user/node-opcua-connect-error/node_modules/node-opcua-client/source/private/client_base_impl.ts:972:27)
    at <anonymous> (/home/user/node-opcua-connect-error/node_modules/node-opcua-client/source/private/client_base_impl.ts:743:29)
    at <anonymous> (/home/user/node-opcua-connect-error/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:665:17)
    at /home/user/node-opcua-connect-error/node_modules/async/dist/async.js:3682:19
    at wrapper (/home/user/node-opcua-connect-error/node_modules/async/dist/async.js:272:20)
    at replenish (/home/user/node-opcua-connect-error/node_modules/async/dist/async.js:442:29)
    at /home/user/node-opcua-connect-error/node_modules/async/dist/async.js:452:13
    at eachOfLimit (/home/user/node-opcua-connect-error/node_modules/async/dist/async.js:478:36)
    at awaitable (/home/user/node-opcua-connect-error/node_modules/async/dist/async.js:212:32)
    at eachOfSeries (/home/user/node-opcua-connect-error/node_modules/async/dist/async.js:814:16)
```

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (x) Linux : version : Ubuntu 22.04.5 LTS (via WSL2)
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

None, I'm trying to connect to an none existing server, to try for infinit retries.

- **node-opcua version:** :
v2.151.0
(still works properly with v2.150.0)

- **Node:**

  `node --version` = v22.13.1

- **Environment**
executed via tsx 4.19.3

erossignon, 2025-04-10:
@wrnotc the issue is fixed in [node-opcua@2.152.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.152.0)

wrnotc, 2025-04-10:
@erossignon yes, I tested it and the issue is fixed in [node-opcua@2.152.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.152.0).

optimistic-optician, 2025-04-17:
@erossignon I can confirm [node-opcua@2.152.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.152.0) is working again flawlessly.