node-opcua/node-opcua #1435: remote server clock doesn't match this computer date

axi92, 2025-04-15:
I just updated the Time on the OPCUA-Server.
Now I have my systemtime (ubuntu 24.04): Tue Apr 15 10:04:54 AM CEST 2025
And the client says:
```
08:05:10.398Z :client_secure_channel_layer   :1218  [NODE-OPCUA-W33]  client : server token creation date exposes a time discrepancy early by 59 minutes 59.329 seconds
 ...                                                remote server clock doesn't match this computer date !
 ...                                                 please check both server and client clocks are properly set !
 ...                                                 server time :2025-04-15T09:05:09.714Z
 ...                                                 client time :2025-04-15T08:05:10.385Z
 ...                                                 transaction duration = early by 0.008 seconds
 ...                                                 server URL = opc.tcp://192.168.3.123:4840 
 ...                                                 token.createdAt  has been updated to reflect client time
```
The client time is calculated kind of wrong.
Do I have to set some kind of ENV var to node or opcua to give it my timezone.

Server is in sync with client:

[image]

erossignon, 2025-04-15:
We use GMT ISO time,not localtime

axi92, 2025-04-15:
What is the workaround then? Is there a way to disable only this warning? Or set an env var for the node process?

erossignon, 2025-04-15:
The warning informs you that there is a discrepancy of one hour between the server and the client

server time :2025-04-15T09:05:09.714Z
client time :2025-04-15T08:05:10.385Z

ChapGPT o1:

According to the OPC UA specification, the practical “maximum allowed” clock skew (time discrepancy) between a Client and Server is 5 minutes. In other words, OPC UA applications are expected to accept certificates and timestamps as valid if their local clock differs by up to five minutes from the remote system’s clock. Beyond this range, time-based checks (for example, during SecureChannel negotiation or certificate validation) will typically fail.

## Where the 5-minute guideline appears in the official specs
You will find this guideline referenced in the OPC UA documentation on security and session establishment. In particular:

**OPC UA Part 4 – Services**, in the sections describing CreateSession and ActivateSession, discusses certificate validation and allowable clock skew.

The exact text varies by revision, but it consistently recommends that vendors tolerate up to a 5-minute difference between the Client’s and Server’s system clocks.

For example, in many vendor interpretations or wrappers of the specification, you will find language such as:

“The Server should allow for a time difference of up to 5 minutes between its system time and the timestamps contained in the Client’s request messages.”

## Why this matters
OPC UA relies on timestamps in tokens and certificates to prevent replay attacks and ensure message freshness. If a Client and Server drift too far out of synchronization, legitimate sessions might get rejected or, conversely, expired sessions might still be accepted erroneously. Hence, 5 minutes is the commonly used upper threshold for that time discrepanc

erossignon, 2025-04-15:
May be the time appear correct on your linux machine but the timezone or Daylight saving is incorrect, 

I would suggest you to consult some Linux forum or contact the support team of your plc manufacturer for advice.

may be timedatectl tool could help you diagnose the issue, 

```
$ timedatectl
               Local time: Tue 2025-04-15 19:21:47 CEST
           Universal time: Tue 2025-04-15 17:21:47 UTC
                 RTC time: Tue 2025-04-15 17:21:47
                Time zone: Europe/Paris (CEST, +0200)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no 
```

axi92, 2025-04-16:
Thanks for trying to help me!
I have the same offset as you from Local to UTC

```
$ timedatectl
               Local time: Wed 2025-04-16 07:24:37 CEST
           Universal time: Wed 2025-04-16 05:24:37 UTC
                 RTC time: Wed 2025-04-16 05:24:37
                Time zone: Europe/Paris (CEST, +0200)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

I guess its the settings on the PLC.
I don't use the time at all on the PLC I can just fix the offset there.

erossignon, 2025-04-16:
THe best is to contact your PLC manufacturer support team .

axi92, 2025-04-23:
I had time to check that again.
Server and client time are off by 6s this should not trigger the warning right? Or does it has to be below 5s?

```
14:00:29.726Z :client_secure_channel_layer   :1218  [NODE-OPCUA-W33]  client : server token creation date exposes a time discrepancy late by 6.142 seconds
 ...                                                remote server clock doesn't match this computer date !
 ...                                                 please check both server and client clocks are properly set !
 ...                                                 server time :2025-04-23T14:00:23.570Z
 ...                                                 client time :2025-04-23T14:00:29.712Z
 ...                                                 transaction duration = early by 0.007 seconds
 ...                                                 server URL = opc.tcp://192.168.3.83:4840 
 ...                                                 token.createdAt  has been updated to reflect client time
{"context":"NestFactory","level":"info","message":"Starting Nest application... ","timestamp":"2025-04-23T14:00:31.100Z"}
```

erossignon, 2025-05-02:
The 5 minutes skew  is mentionned here 

https://reference.opcfoundation.org/GDS/v104/docs/9.5
and here

https://reference.opcfoundation.org/Core/Part6/v104/docs/6.3

erossignon, 2025-05-03:
[node-opcua@2.153.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.153.0) now use a 5 minutes default clock skew, This default can be overriden easily if need be.

[axi92](https://github.com/axi92), please give it a try and confirm

axi92, 2025-05-09:
Thank you that fixed the issue!

closed 2025-05-09 by axi92