node-opcua/node-opcua #1436: No baseDataTypeFactories when using only populate_data_type_manager_104

wrnotc, 2025-04-16:
**Current behavior**

When a server is only using the function populateDataTypeManager104 when using the populateDataTypeManager function (because the server fully implements DataTypeDefinition) the different DataTypeFactories do not have all the necessary baseDataTypeFactories.

**Describe the bug**

When the function `populateDataTypeManager` is called, it calls `populateDataTypeManager103` or `populateDataTypeManager104` or both. 

If `populateDataTypeManager103` is called during the `populateDataTypeManager` call, the dependencies of the different node sets are read from the TypeDictionaries (from the integrated raw schema / integrated bsd file) of the server and the `baseDataTypeFactories` are correctly updated at the end of `populateDataTypeManager103`.

But when only the populateDataTypeManager104 is called all DataTypeFactories (besides the DataTypeFactories for the UA namespace) only have the DataTypeFactories for the UA namespace set as baseDataTypeFactory.

The problem is, that the dependencies are not traced and therefore the baseDataTypeFactories are not updated correctly with the `repairBaseDataFactories` method.

**Expected behavior**

All baseDataTypeFactories should be set correctly.

**Context**
- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (x) Linux : version : Ubuntu 22.04.5 LTS (via WSL2)
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

- **node-opcua version:** :
v2.152.0

- **Node:**

  `node --version` = v22.13.1

- **Environment**
executed via tsx 4.19.3

wrnotc, 2025-04-16:
Added a pull request for a fix:
[PullRequest #1437](https://github.com/node-opcua/node-opcua/pull/1437)

erossignon, 2025-05-03:
This is now fixed in [node-opcua@2.153.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.153.0).

@wrnotc, please give it a try and confirm, thank you for the PR.

closed 2025-05-12 by wrnotc