node-opcua/node-opcua #1440: How to add variables in batches

shizukanaqun, 2025-04-28:
I want to create a grouping of 100,000 variables, but the creation process is very slow. Is there an efficient way to create a large number of data points?
The creation code I am using now is as follows:

const testpoint = namespace.addFolder(rootFolder.objects, { browseName: "testpoint" });

        for(var i = 0; i < 100000; i++) {
            console.log("i = ", i);
            namespace.addVariable({
                organizedBy: testpoint,
                browseName: "point_" + i,
                nodeId: "ns=1;s=testpoint/point_" + i,
                dataType: "Double",
                value: new Variant({ dataType: DataType.Double, value: 1000.0 })
            });
        }

AndreasHeine, 2025-04-28:
doe they change in datatype,amount and or names?

if not create them once and export them as nodeset! loading the nodeset is pretty fast ;)

erossignon, 2025-05-02:
Hi, I confirm a  regression here causing namespace.addVariable on the same node to become uncessarilt slower and slower as the number of children grows.

shizukanaqun, 2025-05-02:
browseName is different，100000 different variables @AndreasHeine

erossignon, 2025-05-03:
This is now fixed in [node-opcua@2.153.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.153.0).

@shizukanaqun,  please give it a try and confirm

shizukanaqun, 2025-05-09:
@erossignon ok
thanks,I will try it

jla, 2025-11-12:
Also the **isFrugal** flag of the AddressSpace makes a huge difference

```typescript
const addressSpace = server.engine.addressSpace;
addressSpace.isFrugal = true;
```