node-opcua/node-opcua #1442: Client internally creates and keeps a session, even if createSession failed.

wrnotc, 2025-04-30:
**Describe the bug**

When you create a client, connect it to a server and then unsuccessfully try to create a session with the createSession or the createSession2 method, the client will have an additional session in the OPCUAClient._session Array, even though the createSession did run into an error.

As a result of this, if e.g. the client looses connection to the server and then reestablishes it, the client tires to repair/reestablish the session, even though it never was a valid session before.

**Unit test**

Unit test to reproduce the bug, including comments explaining the bug:
```mts
import {
    OPCUAServer,
    OPCUAServerOptions,
    OPCUAClient,
    MessageSecurityMode,
    SecurityPolicy,
    OPCUAClientOptions,
    EndpointWithUserIdentity,
    UserTokenType,
    ClientSession
} from "node-opcua"
import should from "should";

const serverOptions: OPCUAServerOptions = {
    port: 4840,
    securityPolicies: [SecurityPolicy.None],
    securityModes: [MessageSecurityMode.None],
    allowAnonymous: false
};

const clientOptions: OPCUAClientOptions = {
    endpointMustExist: false,
    connectionStrategy: {
        maxRetry: 2,
        initialDelay: 250,
        maxDelay: 500,
    },
};

const endpoint: EndpointWithUserIdentity = {
    endpointUrl: "opc.tcp://localhost:4840",
    userIdentity: {
        type: UserTokenType.Anonymous
    }
}

const describe = require("node-opcua-leak-detector").describeWithLeakDetector;
describe("issue_XXX", function (this: any) {
    this.timeout(10 * 1000);
    it("session which cannot be created should not be present in the client", async () => {
        const server = new OPCUAServer(serverOptions);
        try {
            // start server
            await server.initialize();
            await server.start();

            // start client
            const client = OPCUAClient.create(clientOptions);
            try {
                await client.connect(endpoint.endpointUrl);
                let failed = false;
                let session: ClientSession | null = null;
                try {
                    // Will always fail, because of invalid authentication.
                    // Note: Bug is the same for OPCUAClient.createSession2() method.
                    session = await client.createSession(endpoint.userIdentity); 
                }
                catch {
                    failed = true
                } 
                finally {
                    should(failed).be.true(); // createSession successfully failed 
                    should(session).is.null(); // therefore no session should exist as return value
                    should(client).have.property('_sessions').with.lengthOf(0); // and not in the client
                }
            }
            finally {
                await client.disconnect();
            }

        } finally {
            // shutdown server
            await server.shutdown();
        }
    })
});
```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Start this server:
```mts
import {
    OPCUAServer,
    OPCUAServerOptions,
    MessageSecurityMode,
    SecurityPolicy
} from "node-opcua"

const serverOptions: OPCUAServerOptions = {
    port: 4840,
    securityPolicies: [SecurityPolicy.None],
    securityModes: [MessageSecurityMode.None],
    allowAnonymous: false
};

// to force exit
process.on('SIGINT', () => {
    console.log("Server terminated without proper shutdown.")
    process.exit(0);
});

const server = new OPCUAServer(serverOptions);
await server.initialize();
await server.start();

console.log("Server started.")
```
2. Start this client:
```mts
import {
    OPCUAClient,
    OPCUAClientOptions,
    EndpointWithUserIdentity,
    UserTokenType
} from "node-opcua-client"
import EventEmitter from "node:events";

const clientOptions: OPCUAClientOptions = {
    endpointMustExist: false,
    connectionStrategy: {
        maxRetry: 2,
        initialDelay: 250,
        maxDelay: 500,
    },
};

const endpoint: EndpointWithUserIdentity = {
    endpointUrl: "opc.tcp://localhost:4840",
    userIdentity: {
        type: UserTokenType.Anonymous
    }
}

// to force exit
process.on('SIGINT', () => {
    console.log("Client terminated without proper shutdown.")
    process.exit(0);
});

const client = OPCUAClient.create(clientOptions);
try {
    await client.connect(endpoint.endpointUrl);
    client.on("connection_lost", () => console.log("Lost connection."))
    client.on("after_reconnection", () => console.log("Reestablished connection."))
    try {
        // Will always fail, because of invalid authentication.
        // Note: is the same for OPCUAClient.createSession2() method.
        await client.createSession(endpoint.userIdentity);
    }
    catch {
        console.log("Create Session failed as intended.")
    }
}
catch {
    console.log("Unexpected error.")
}
```
3. Terminate the Server via <kbd>Ctrl</kbd> + <kbd>C</kbd>
4. Start a new instance of the Server.
5. Now the client reconnects to the server and, because it internally still has a session, it will try to reestablish/repair the session, even though it was not successful at first.

Log then looks similar to:
```
Create Session failed as intended.
Lost connection.
Reestablished connection.
11:12:42.482Z :reconnection                  :527   session restoration has failed! err = Cannot find ANONYMOUS user token policy in end point description ns=1;g=09DD93AB-34A7-0E50-5230-7D2B235D932D  => Let's retry
11:12:42.483Z :reconnection                  :535   ... will retry session repair... in 2000 ms
11:12:44.483Z :reconnection                  :541   Retrying session repair...
11:12:44.488Z :reconnection                  :527   session restoration has failed! err = Cannot find ANONYMOUS user token policy in end point description ns=1;g=398E7E2F-741F-E740-0A6B-4BD834B7ACFA  => Let's retry
11:12:44.488Z :reconnection                  :535   ... will retry session repair... in 2000 ms
11:12:46.489Z :reconnection                  :541   Retrying session repair...
11:12:46.493Z :reconnection                  :527   session restoration has failed! err = Cannot find ANONYMOUS user token policy in end point description ns=1;g=61510C28-AACA-4559-8126-AEB0CBC6432C  => Let's retry
11:12:46.493Z :reconnection                  :535   ... will retry session repair... in 2000 ms
```

**Expected behavior**

If the session cannot be established successfully it shall not be present in the OPCUAClient object.

**Screenshots**

**Context**
- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (x) Linux : version : Ubuntu 22.04.5 LTS (via WSL2)
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

- **node-opcua version:** :
v2.152.0

- **Node:**

  `node --version` = v22.13.1

- **Environment**
executed via tsx 4.19.3

erossignon, 2025-05-03:
This is now fixed in [node-opcua@2.153.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.153.0).

@wrnotc, please give it a try and confirm

wrnotc, 2025-05-09:
I can confirm, that Version 2.153 fixes the issue. 
@erossignon thank you for fixing it.